micropython/micropython #4030: run-tests: Test name duplication masks the failures

pfalcon, 2018-08-07:
E.g. there's basics/string_format.py and float/string_format.py . If basics/string_format.py failed, string_format.py.exp and string_format.py.out will be produced in run-tests's dir. But then if float/string_format.py didn't fail, it'll remove string_format.py.exp and string_format.py.out created in the previous step, leaving a user to wonder what happens.

stinos, 2018-08-07:
Ah yes have been bitten by this once as well. What would you suggest? Use subdirectory name for distinction like basics_string_format.out and float_string_format.out? Or put .out and .exp next to the .py?

pfalcon, 2018-08-07:
> like basics_string_format.out and float_string_format.out?

Yep, that's an obvious way to address it. More involved one is to actually ensure that test names are unique, but that would require adding a code to run-tests to enforce it, a bit boring.

> Or put .out and .exp next to the .py?

That would be the least preferrable option IMHO, the whole idea that failure cases are easily accessible, without need to hunt for them in multiple subdirs.

dpgeorge, 2018-08-10:
Should be addressed by ca0d78cebb20ed67f78491a68b02272aba2ecd13

closed 2018-08-10 by dpgeorge