micropython/micropython #4035: conversation float to int on MCU

ondratu, 2018-08-08:
Hi, i see some magic int / float operations on MCU. I want to get timestamp aligned to minutes on ESP8266. Here is output from console:
``` python
>>> now = time.time()
>>> int(now/60)*60, now
(587021520, 587021509)
```

I test in on your web unicorn and here is output:
``` python
>>> now = 587021509                                                            
>>> int(now/60)*60, now                                                        
(587021520, 587021509)
```

Unix port works OK:
``` python
now = 587021509 
>>> int(now/60)*60, now
(587021460, 587021509)
```

When i do this more-time on ESP, the first number is still same:
``` python
>>> now = time.time()
>>> int(now/60)*60, now
(587021760, 587021738)
>>> now = time.time()
>>> int(now/60)*60, now
(587021760, 587021759)
>>> now = time.time()
>>> int(now/60)*60, now
(587021760, 587021767)
>>> int(now/60)*60, now
(587021760, 587021767)
>>> now = time.time()
>>> int(now/60)*60, now
(587021760, 587021837)
>>> now = time.time()
>>> int(now/60)*60, now
(587021760, 587021901)
```

firmware on ESP8266:  esp8266-ota-20180511-v1.9.4.bin
```
>>> sys.version
'3.4.0'
>>> sys.implementation
(name='micropython', version=(1, 9, 4))
>>> sys.platform
'esp8266'
```
Later i would to test it on STM32F4 Discover and on ESP32.

dpgeorge, 2018-08-08:
The esp8266 only has 30 bits of precision for a float, so there is loss of precision here.  Instead, try to use truncating integer division: `int(now // 60) * 60`

robert-hh, 2018-08-08:
It's most likely caused by the limited significant digits in a float operation. You should get the expected result by `(now // 60) * 60` instead of `int(now / 60) * 60`.

ondratu, 2018-08-08:
Thanks, that works.

This could be documented in differences that it works like C not like CPython. Resp. that not works like C, here we could get overflow number, not still the same.

closed 2018-12-04 by dpgeorge