micropython/micropython #4041: How to add a new target about   stm32f412 or 407  or  different pins number chip

zouzhe1, 2018-08-09:
I dont find a turiual about   add a new target  describle.
where ?
please help me

I see the  pin define  not design in one  file,  it have been design some file....
this  design is bery bad!

zouzhe1, 2018-08-09:
yes，it is this。I want to add a STM32 microcontroller that is not already supported (see stm32/boards directory)。My board is  STM32F412DISC。or if i have an other chip(stm32)   ,How should i do it?

rolandvs, 2018-08-09:
What you need to do to GET STARTED:

Look in my repo at branch `rolandvs/micropython/board_STM32F412DISC` for the files you need.

You have to add/check the `stm32f412_af.csv`
You have to add/check the `stm32f412.ld` for the memory size of flash, ram, etc.
You have to adapt `mpboardconfig.h` for the correct GPIO pins.

Then it might be necessary to adapt some settings in the code. Best for you is to browse through it and see how it is done if you run into problems.

I have made a few changes already... It needs work: the stuff compiles, however you can start looking at `flashbdev.c` to add the processor or configure internal flash of (mpboardconfig.h)

rolandvs, 2018-08-09:
BTW: the STM32F407 is almost identical to the STM32F405. So supporting that one is very easy. You can copy from the STM32F4DISC (which uses an STM32F407).

closed 2018-12-04 by dpgeorge

sfynxu, 2019-02-17:
Ok. How about STM32F3 Discovery board?
Where should I start?

chrismas9, 2019-02-17:
There is a discussion on the forum and work in progress on the F3.

https://forum.micropython.org/viewtopic.php?f=12&t=3426&sid=41042a07bb1d7fff15f4b2476999a888