micropython/micropython #4042: ESP32 RTC memory wrong location

Lisa999, 2018-08-09:
From the esp-idf's rtc.h file:

> /**************************************************************************************
  *                                       Note:                                       *
  *       Some Rtc memory and registers are used, in ROM or in internal library.      *
  *          Please do not use reserved or used rtc memory or registers.              *
  *                                                                                   *
  *************************************************************************************
  *                          RTC  Memory & Store Register usage
  *************************************************************************************
  *     rtc memory addr         type    size            usage
  *     0x3ff61000(0x50000000)  Slow    SIZE_CP         Co-Processor code/Reset Entry
  *     0x3ff61000+SIZE_CP      Slow    4096-SIZE_CP
  *     0x3ff62800              Slow    4096            Reserved
  *
  *     0x3ff80000(0x400c0000)  Fast    8192            deep sleep entry code
  *
  *************************************************************************************
  *     RTC store registers     usage
  *     RTC_CNTL_STORE0_REG     Reserved
  *     RTC_CNTL_STORE1_REG     RTC_SLOW_CLK calibration value
  *     RTC_CNTL_STORE2_REG     Boot time, low word
  *     RTC_CNTL_STORE3_REG     Boot time, high word
  *     RTC_CNTL_STORE4_REG     External XTAL frequency
  *     RTC_CNTL_STORE5_REG     APB bus frequency
  *     RTC_CNTL_STORE6_REG     FAST_RTC_MEMORY_ENTRY
  *     RTC_CNTL_STORE7_REG     FAST_RTC_MEMORY_CRC
  *************************************************************************************
  */

BUT... the RTC slow memory is NEVER used:
`#define MEM_MAGIC           0x75507921
/* There is 8K of rtc_slow_memory, but some is used by the system software
    If the USER_MAXLEN is set to high, the following compile error will happen:
        region `rtc_slow_seg' overflowed by N bytes
    The current system software allows almost 4096 to be used.
    To avoid running into issues if the system software uses more, 2048 was picked as a max length
*/
#define MEM_USER_MAXLEN     2048
RTC_DATA_ATTR uint32_t rtc_user_mem_magic;
RTC_DATA_ATTR uint32_t rtc_user_mem_len;
RTC_DATA_ATTR uint8_t rtc_user_mem_data[MEM_USER_MAXLEN];
`
The buffer rtc_user_mem_data is never pointing to the RTC slow memory, so after a reboot the contents of the buffer is clean...

MrSurly, 2018-11-21:
@manningt -- I think you wrote this?

manningt, 2018-11-21:
According to the ESP-IDF [deep-sleep docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/deep-sleep-stub.html?highlight=RTC_DATA_ATTR), variables that are declared with RTC_DATA_ATTR are stored in RTC memory.  That memory is only persisted/restored after deep_sleep.  It's erased on reset.  I do not know whether:
1. the ESP-IDF code copies RTC memory to the rtc_user_mem_data buffer on wake-up from deep-sleep, or
2. the rtc_user_mem_data is a pointer to RTC memory.

I do know that the rtc_user_mem_data is preserved through a deep sleep.  I have working code that uses it.

jonnor, 2024-07-27:
6 years later many updates have been done to the ESP32 port. Is this issue still present?

Lisa999, 2024-07-27:
No longer relevant…

closed 2024-07-27 by robert-hh