micropython/micropython #4043: [STM32] moving the "*.ld" files into the specific board directories

rolandvs, 2018-08-09:
Dear all,

We have seen some great developments to add things like native Ethernet, SDRAM, etc. However:
- With the latest addition of the SDRAM driver the "generic" `stm32f429.ld` file is changed with a special entry for the SDRAM
- When using the native Ethernet (RMII interface of chip itself) the Ethernet buffers are added to the `ld` file
- When using some specific memory layouts the `ld` file will be different

Place in case: NUCLEO_F429ZI and STM32F429DISC both sharing the `stm32f429.ld` file. In this case the additional line with the SDRAM info doesn't conflict, but with the Ethernet controller some RAM space gets allocated as buffers for Ethernet instead of global RAM usage.

So I would like to make a case to move the `*.ld` files, *for at least the boards with added support for Ethernet and SDRAM*, into the specific board directories. It is a small change in the file `mpconfigboard.mk` file.

aykevl, 2018-08-09:
You can avoid duplication and use multiple linker scripts per board, as has been done by the nrf port.

Defining a port-specific variable:
https://github.com/micropython/micropython/blob/master/ports/nrf/boards/microbit/custom_nrf51822_s110_microbit.ld
Using it on nrf51 boards, with varying filesystem sizes:
https://github.com/micropython/micropython/blob/master/ports/nrf/boards/s110_8.0.0.ld#L6

dpgeorge, 2019-06-24:
Boards are free to use their own complete linker script and not rely on any of the common/generic ones; see eg PYBD_SF2.  So that's an option and should be used when a board has special requirements.

Otherwise, I don't see any issue keeping the common/generic ld files in `stm32/boards/`.  It doesn't hurt to have an SDRAM line in the MEMORY section if it's not used.  And for Ethernet, the buffers are defined in `eth.c` and only included when ETH is enabled in `mpconfigboard.h`.

closed 2019-06-24 by dpgeorge