micropython/micropython #4047: Add native linked-list data-structure

adritium, 2018-08-12:
I'm creating an interface that in uPy that will create many data objects which the user will collect into a container to be used later; for now I'm using a `list`.

```Python
meas_list = []
for _ in range(1000):
    meas_list.append(get_new_data())
```
The problem is that a `list` keeps its items contiguous in memory which is fine for O(1) indexing but in many cases, you don't need the O(1) indexing that a `list` provides.

In embedded environments, we all know that heap memory fragmentation is a reality and using a `list` for the case of aggregating data objects will lead to a fragmentation error sooner than is actually necessary.

It would be very useful to add a native implementation of a linked list.

peterhinch, 2018-08-13:
I don't think Python lists are contiguous in RAM: it's hard to see how they could be given that the contents of an entry are of arbitrary size. Arrays, with predefined contents, are contiguous.

Given that lists aren't contiguous why would you expect them to perform worse than a linked list in terms of fragmentation?

stinos, 2018-08-13:
For MicroPython lists the pointers to objects stored are in contiguous memory though.

But even then it's really not as simple as 'using list means fragmentation problem'. For the example provided you'd preallocate and there would be no problems. Or else you'd really have to measure if the problems come from the list: it's allocation policy is to double the storage whenever needed and it starts with 4 entries. So after the example you show the list reallocated a couple of times to match it's need for 1024 * sizeof(mp_obj_t) of space. So the previous, now freed blocks (512/256/128/...), are all fairly large in size so it really depends on the other operations needing memory if those blocks are usable or not and hense whether fragmentation because of the list is an actual problem. Opposite, but depending on the actual application can be equally problematic, is a linked list: it will happily use small heap blocks all over the place but that also means once you need a big chunck you might get in trouble.

adritium, 2018-08-13:
> once you need a big chunck you might get in trouble.

With both `list` and `linked-list` you need to allocate space for the new obj but with `list`, the collection of `mp_obj_t` must be kept in contiguous memory. If O(1) indexing is not a requirement, you can use `linked-list` and remove the jitter from reallocation.

While the reallocation time for `list` is O(1) _amortized_, you'd still have jitter because of the reallocation.

stinos, 2018-08-14:
You could always code the linked list in Python. Might be a bit less performant but memory/allocation results will be similar enough to a native implementation.

dpgeorge, 2022-03-30:
Closing due to inactivity.

closed 2022-03-30 by dpgeorge