micropython/micropython #4053: Clearing interned string pool . ( for memory optimize )

curlyz, 2018-08-15:
Hi . 
I am currently having problem with interned string . 
I have a file that I only run once . The file is hugh and having lots of identifiers . As the result , after the file is executed , the qstr_info(1) state that 12K of RAM is not freed. 
Is there anyway that I can telling the compiler that the identifiers shouldnt be interned ? 
What I want is to run the file and clean everything that is not used . 
Also , the '`micropython.qstr_info(1)`' return None , is there other way to parse the info ?
Thanks .

adritium, 2018-08-15:
Post a portion of the file.

Not sure if this matters: are strings interned at compile-time (using `MP_QSTR_`) or during runtime?

dpgeorge, 2018-08-17:
> Is there anyway that I can telling the compiler that the identifiers shouldnt be interned ?

Not really.  All identifiers are interned so the compiler can efficiently compare variable names.  Otherwise it would need to store the string data itself and use strcmp (instead of just storing a number and comparing numbers for equality).

> The file is hugh and having lots of identifiers .

A way to work with large scripts is to precompile them to .mpy files using mpy-cross.  Then import the .mpy file.  Only the identifiers that are exported from the script would be interned; local variables in functions will not (but argument names will).

peterhinch, 2018-08-18:
@curlyz There is [pyminifier](https://liftoff.github.io/pyminifier/) which can compress Python code without the need to obfuscate your source. But I see no reason to use it with MicroPython given the availability of mpy-cross and frozen bytecode.

curlyz, 2018-08-18:
@peterhinch  I am trying to use mqtt protocol to deploy a webpage where user can upload and execute there code with support for async (your library is so cool!) , everytime user upload a new code , the previous code will get deleted and all the tasks associated with it will be cancelled .
Such application require cleaning up variables , tasks.... to restore the original heap to the system.
So far , it works great , but the more code user upload , the intern pool get bigger. Since I can't rely on resetting the board to clear the pool (because it will take time to get connected again and also there are no way you can parse the qstr_info ).

adritium, 2018-08-18:
> everytime user upload a new code , the previous code will get deleted and all the tasks associated with it will be cancelled .

@peterhinch @dpgeorge 

Now it's clearer: the code uploading is done _after_ the firmware is built so you cannot use `mpy-cross` to create frozen bytecode. Furthermore, he doesn't want to delete a specific `qstr` but ALL the `qstr`s created thus far.

This is trivial: add `void qstr_reset_pool()` that will delete all the pools except `CONST_POOL`.

peterhinch, 2018-08-19:
This seems to be related to [issue 2280](https://github.com/micropython/micropython/issues/2280).

adritium, 2018-08-19:
@peterhinch 

It may be related but in this issue there's no workaround because the @curlyz cannot control what code the user uploads

> everytime user upload a new code

The only solution is to clear the qstring pool.

adritium, 2018-08-29:
@curlyz are you going to implement this?

I'm not going to write code but I'll be happy to brainstorm with you.

adritium, 2018-08-30:
That's not going to work with current code.

Whenever a new `qstr` is added, it is protected from garbage collection due to the chain of root pointers starting with `qstr_pool_t *last_pool;`. So to remove a given `qstr` you would have to remove it from `const byte *qstrs[];` in the struct

```C
typedef struct _qstr_pool_t {
    struct _qstr_pool_t *prev;
    size_t total_prev_len;
    size_t alloc;
    size_t len;
    const byte *qstrs[];
} qstr_pool_t;
```

The current `find_qstr` assumes the qstrings are in increasing and contiguous order in terms of `qstr q`. This scheme makes lookup super-fast so we should definitely maintain it. 

To delete a qstring this is necessary (but not sufficient): 

1. set the entry in `const byte *qstrs[];` to zero
2. `memcpy` the remaining qstrings in the array downward by 1
3. set the previous "top" entry in `const byte *qstrs[];` to `0`.

This isn't sufficient because of `MP_STATE_VM(qstr_last_chunk)`.
This is pre-allocated in chunks so that everytime you add a qstring you do not need to call `malloc`. That's fine but the problem is that this pointer is forgotten on purpose because it's assumed that qstrings never get deleted - see `qstr_from_strn()` - so this memory can never be reclaimed.

To fix this, modify `qstr_pool_t` so that it has its own `byte * chunk`.

dhylands, 2018-08-30:
You can't just move the qstrs down by 1, unless you also adjust all of the places in the code which use the qstr indicies corresponding to the qstrs you just moved.

adritium, 2018-08-30:
@dhylands yes I realized that.

adritium, 2018-08-30:
Let's back up a bit
The operations on qstrings are

1. create/insert
2. lookup
3. remove
4. find

curlyz, 2020-06-28:
@adritium I made a dirty hack. Just build a simple compiler that look for all strings and combine them into an variable, sync this is long string, it wont get interned. Use python tokenize library to substitute all strings with a function call.
```python
# Normal case
a = "hello"
b = "world"

# Converted
qstr = "5Hello5World" # 5 is length of string
a = f(0)
b = f(6) # 6 is the address of that string.

# Add this caller on top
def f(position):
  length = int(qstr[position])
  return qstr[position:position+length]

# You get the idea.
```

closed 2020-06-28 by curlyz

bhcuong2008, 2021-06-06:
Hi,

I'm also facing this. QSTR helps save memory and faster operations, but waste memory if not using it any more. Could we completely manage QSTR pool, delete it if not any reference to it,..

Otherwise, the only way to clear the pool is reset MPY, using CTRL+D.

Thanks.