micropython/micropython #4056: Adding units to variable names in docs

liudr, 2018-08-16:
This shouldn't be a hard thing to implement when appropriate. Replace 'timeout' with 'timeout_ms' for machine.UART docs for instance, doesn't need any modification to the source code. But the end result is much more clear docs than right now. I also recommend adding timeout_ms=xxxx in the first snippet in the doc to show users how to use it.

This is especially important since CPython on PCs use second-based timeout in pyserial and it's not well emphasized in their docs either. So assuming many new MicroPython users have some PC Python backgrounds, like myself, they will assume the timeout would be in seconds only to find that timeout was 'not working' because they timed out in 10 milliseconds instead of 10 seconds. Maybe a note can be added to chapter 6 "differences" due to microcontroller convention or something.

Other less critical ones are time.sleep(). Since there are milliseconds and microseconds counterparts, why not adding an alias such as time.sleep_s()? Just like magic numbers, function/variable without units when there might be ambiguity should be highly discouraged.

In all my years learning programming, units aren't emphasized. I think it's a mistake. Hope you concur, with a fellow physicist's point of view. Thanks.

peterhinch, 2018-08-17:
I would suggest raising a PR improving the docs rather than making changes which break existing code. In particular the `time` library is standard Python so changing `time.sleep()` would be a non-starter. Aliases avoid breaking code but clutter the namespace.

jonnor, 2024-09-19:
As mentioned, documentation improvements should be made rather than breaking API changes.

closed 2024-09-19 by jonnor

liudr, 2024-09-19:
Well, you choose to do these aliasing "clutter" and other things not because it is easy, but because it is hard. Since the beginning of language, doing the right things has often been dismissed with "we've always done it that other way so beat it!".