micropython/micropython #4060: ESP32 wake on touch

tahouse, 2018-08-17:
Hi there,

I'm running MicroPython v1.9.4-409-g434975def and have successfully used a Pin irq to wake from deep sleep and can confirm the wake reason:

```
import machine
from machine import Pin, TouchPad, deepsleep
import time

wake = Pin(14, mode = Pin.IN, pull = Pin.PULL_DOWN)
wake.irq(trigger=Pin.WAKE_HIGH, wake = machine.DEEPSLEEP)

time.sleep(5)
deepsleep(5*1000)
```
after bringing Pin 14 line up to 3V3
```
Reset cause:  DEEPSLEEP_RESET
Wake reason:  PIN_WAKE/EXT0_WAKE
```

Now I'm trying to do the same but with a TouchPad input:

```

from machine import Pin, TouchPad, deepsleep
import time
import esp32

wake = Pin(14, mode = Pin.IN)
touch = TouchPad(wake)
touch.config(500)
esp32.wake_on_touch(True)

time.sleep(5)
deepsleep(5*1000)
```
(after 5 seconds and touching/untouching wire to Pin 14)
```Reset cause:  DEEPSLEEP_RESET
Wake reason:  TIMER_WAKE```

The range of my touch.read() values are around 700 when untouched, and down to around 100 when touched. I've tried adjusting the touch.config to various settings but I cannot get my ESP32 to wake from touch.. only from timer in this mode.

Lastly,
I've tried using wake_on_ext1
```
import machine
from machine import Pin, TouchPad
import time
import esp32
wake = Pin(14, mode = Pin.IN, pull = Pin.PULL_DOWN)
# wake.irq(trigger=Pin.WAKE_HIGH, wake = machine.DEEPSLEEP)
esp32.wake_on_ext1(pins = [wake], level = Pin.WAKE_HIGH)

time.sleep(5)
deepsleep(5*1000)
```
But this resets immediately regardless of whether I bring Pin 14 high or not (not in touch mode, mind you).
```
Reset cause:  DEEPSLEEP_RESET
Wake reason:  EXT1_WAKE
```

I've browsed through the esp32 module code, but I'm not sure if any of the wake modes work?

Any ideas?

peterhinch, 2018-08-18:
Your query will find a larger audience in [the forum](https://forum.micropython.org/). This is the MicroPython bug tracker.

tahouse, 2018-08-20:
Hi Peter, 
Thanks for the response, I'll crosspost to the forum. 

That said, I haven't found any documentation available for the wake_on_touch method or the esp32 module at all in the micropython repository, so in the very least this should be filed as a **documentation** issue.

manningt, 2019-05-05:
To get wake on touch to work, the following ESP-IDF API needs to be called:
`    touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);
`
The following ESP-IDF API can be used to reduce the power consumption of the touch pins during deep sleep:
`touch_pad_set_meas_time(0xF000, 0x4000);`

After putting both of these in the `esp32_wake_on_touch` routine in modesp32.c, then wake-on-touch worked.

Note: `touchpad.config()` needs to be performed _after_ esp.wake_on_touch(True).

I got touch-on-wake working by following the sequence of calls shown in the [deep_sleep_example_main.c](https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/deep_sleep_example_main.c) in the ESP-IDF repository.

Should I create a branch and pull request to add these functions to esp.wake_on_touch ?

dpgeorge, 2019-05-07:
> Should I create a branch and pull request to add these functions to esp.wake_on_touch ?

Yes, please do.

dpgeorge, 2019-05-28:
Wake on touch was fixed in 887a6712c2a302bd0bdeceb8ee9bc33a6de64dae