micropython/micropython #4062: External interrupts on a Pin doesn't report the line number in callback

rhubarbdog, 2018-08-18:
I'm following the documentation at http://docs.micropython.org/en/v1.9.3/pyboard/library/pyb.ExtInt.html
It states that the callback registered by `pyb.ExtInt` has one parameter `line number`.  I wrote a little proof of concept program wondering if the line number reported would be that of the `pyb.ExtInt` command or somewhere later in a while loop that the program waits for input.

robert-hh, 2018-08-18:
Are you expecting the python source line number? As far as I understand, the documentation talks about the port line number which causes the interrupt.

rhubarbdog, 2018-08-18:
my bad, yes i was expecting a source code line number, but now i've read the docs again it is the port line

closed 2019-05-21 by dpgeorge