micropython/micropython #4064: Question: `memoryview.cast` in MicroPython

Peque, 2018-08-19:
It seems `memoryview` has no `cast()` method in MicroPython, has it?

Is there another way to create a memoryview out of an `array.array('h', array_size)` that results in a bytearray-like memory view without making a copy of the buffer?

In code, for better clarity:

```python
import array

size = 64
a = array.array('h', range(size))
m = memoryview(a).cast('B')
assert len(m) == 2 * size
```

adritium, 2018-08-19:
If you have a question, post it on the forum https://forum.micropython.org/index.php

peterhinch, 2018-08-19:
See [the docs](http://docs.micropython.org/en/latest/pyboard/library/uctypes.html). Hint: `uctypes.addrssof()` and `uctypes.bytearray_at()`. But as @adritium has pointed out, this is the MicroPython bug tracker.

Peque, 2018-08-19:
@adritium @peterhinch Thanks for your help! :blush:

I opened this issue as I was not sure if this could be considered a bug or unimplemented feature (i.e.: the behavior is different from Python's, where the `.cast()` method exists). Feel free to close it if this is not really a MicroPython's issue.

jonnor, 2024-10-02:
I do not immediately see how to implement a cast with uctypes. Just getting a bytearray can be done, but I cannot see how to go another way, say interpret a bytearray as int16.