micropython/micropython #4065: implement bit_length() for mp_type_int

adritium, 2018-08-19:
[int.to_bytes(size, byteorder)](http://docs.micropython.org/en/latest/pyboard/library/builtins.html?highlight=to_bytes#int.to_bytes) requires one to know how many bytes - `size` - the output is but `int.bit_length()` is not implemented.
```Py
>>> (1024).to_bytes(2, byteorder='big')
b'\x04\x00'
>>> (1024).to_bytes(10, byteorder='big')
b'\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00'
>>> (-1024).to_bytes(10, byteorder='big', signed=True)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xfc\x00'
>>> x = 1000
>>> x.to_bytes((x.bit_length() + 7) // 8, byteorder='little')
b'\xe8\x03'
```

dmazzella, 2018-08-20:
Attached a patch file that implement int.bit_length()

[int_length.zip](https://github.com/micropython/micropython/files/2303186/int_length.zip)

adritium, 2018-08-20:
If you went to the trouble to implement it, why not submit a pull-request?

C4, 2019-03-07:
+1 for getting this added in.

rkagerer, 2023-02-26:
Has this still not been merged?

dpgeorge, 2023-02-26:
> Has this still not been merged?

There is no pull request for this feature, so nothing to review and merge.

dmazzella, 2023-05-31:
Ciao Massimo,
I am attaching an updated version of the patch file to make `int.bit_length` work.

this patch enable `MICROPY_INT_BIT_LENGTH` into the file `ports/unix/variants/mpconfigvariant_common.h` and unix port can be used for test.

Try it and let me know if it works correctly.

this is a test for him, create the file `int_bit_length.py` into the folder `tests/basics/` and paste:

```
# tests int.bit_length

n = -37
print(n.bit_length())

n = 1024
print(n.bit_length())

n = -1024
print(n.bit_length())

print((2048).bit_length())

print((-2048).bit_length())

```

If everything works correctly I will proceed to create a PR for this

Greetings,
D.
[int_bit_length.patch](https://github.com/micropython/micropython/files/11614328/int_bit_length.patch)

massimosala, 2023-05-31:
Ciao Damiano

What about the possible optimization in the mpn/mpz code path?

---

For the testing: sorry, I just started developing on MP and I still haven't tried to install the toolchain and compile a firmware.