micropython/micropython #4067: Add function to move mp_obj to/from readonly memory

adritium, 2018-08-21:
This is from a forum post

https://forum.micropython.org/viewtopic.php?t=5171

RAM is very valuable on most things that run micropython. 

Currently the only way to put something in readonly memory is to freeze it and build it as part of the firmware but for some applications, that workflow is impossible. 

For example: you deploy your micropython module and users can upload .py files to run. In this scenario it’s possible to invoke the C compiler. 

Sometimes an mp_obj, like array, never changes after a certain point. 
It would be nice to be able to move that array to readonly memory. 

Implementation

A copy of the object must be made in flash. A deep copy. 

(I’m mobile and I’ll continue tomorrow)

peterhinch, 2018-08-21:
Even if you freeze an object as Python source you have to take precautions to prevent it from being copied to RAM on import.

A mutable object like an array will be copied to RAM because the runtime doesn't 'know' if code might modify it. As a minimum, data must be stored as in immutable objects, and if subsets are to be accessed it may be necessary to use `memoryview` instances to suppress copying.

adritium, 2018-08-21:
> A mutable object like an array will be copied to RAM because the runtime doesn't 'know' if code might modify it.

This is a good start to implement this feature.
Firstly, this would have to be a micropython-specific feature since normal Python has no concept of readonly vs readwrite memory.

We would not be able to use `memoryview` to implement this because that requires that the `array` already exists . . . and it would need to exist in RAM because that's the only way an `array` can be created.

We would have to create new data types that **know** whether they're currently in readonly or readwrite memory so that they can control modification of the data structure.

Example:

```Py
# create a the list in RAM
mylist = list_freezable([1, 2, 3, 5, 34, 2])
# do some operations
mylist.append(5)

# freeze the list
mylist.freeze()
# list is now in readonly memory and RAM has been liberation

# operation will result in error: "list is frozen"
mylist.append(55)
```

peterhinch, 2018-08-21:
That looks good.

adritium, 2018-08-21:
> Yes I agree. It was never my intention to come up with a method to completely avoid the VM placing the object in RAM initially

I now disagree with my statement; initially creating the entire object in RAM reduces the usefulness of object-freezing.

Let's say you have an application that generates 1,000 uint32_t coefficients on the PC and you have to have access to all of them while running in uPy.

Ideally, you'd do it by generating the following Python line into a `.py` file.

```Py
Coeffs = [32, 453, 1, 4, 3412, 12, 213, 123, ...]
```
But merely loading this `.py` file would consume huge amounts of RAM: each character is 1-byte and each number can be up to 9 characters -> up to 9,000 bytes to simply load this `.py` file into RAM.
To eliminate the RAM needed to simply store the file in memory, you would save the `.py` file to readonly memory like flash.

But once the this line is executed, you'd consume 4,000 bytes to merely hold the list in memory.

What about `freezable_list`?

```Py
Coeffs = freezable_list(32, 453, 1, 4, 3412, 12, 213, 123)
```

If, as I said, initially the object is allowed to live in RAM, the usefulness is greatly reduced.
The number of coeffs you could load from the PC is limited by how much RAM you have.
Yes, you could immediately freeze the object but there's still a limit.

Ideally, the object would never exist in RAM in the first place.

**Just realized something**
If you compile from `.py`->bytecode on the embedded module, it's impossible to eliminate intermediate RAM from being used.

We saw why that's true if you load in a `.py` file in RAM to be compiled but it's also true even if you place the `.py` file in readonly memory.

Why? Because the resulting bytecode will be in RAM.
This `.py` _line_, whether in RAM or in flash . . .

```Py
Coeffs = freezable_list(32, 453, 1, 4, 3412, 12, 213, 123, 324, 234)
```

. . . will result in this bytecode.

```C
STATIC const byte bytecode_data_frozen_ar__lt_module_gt_[46] = {
    0x0b, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_frozen_array & 0xff, MP_QSTR_frozen_array >> 8,
    0x00, 0x00, 0xff,
    0x1b, MP_QSTR_freezable_list & 0xff, MP_QSTR_freezable_list >> 8,
    0xa0, 
    0x14, 0x83, 0x45, 
    0x81, 
    0x84, 
    0x14, 0x9a, 0x54, 
    0x8c, 
    0x14, 0x81, 0x55, 
    0x14, 0x80, 0x7b, 
    0x14, 0x82, 0x44, 
    0x14, 0x81, 0x6a, 
    0x64, 0x0a, 
    0x24, MP_QSTR_Coeffs & 0xff, MP_QSTR_Coeffs >> 8,
    0x11, 
    0x5b, 
};
```

POST NOT DONE

adritium, 2018-08-21:
Given the limitation the Py stack introduces, I am forced to reconsider the API for `frozen_*`.

Expanding the size of a structure in flash is time-consuming and should be avoided though technically it's doable.

Therefore, for a given `frozen_*` object the user would have to specify the total size implicitly and upfront: if it's `frozen_array`, they specify the count and format code.

```Py
# Allocate enough memory and allocate it in readonly area
coeffs = frozen_array(1000, 'l', 'readonly')
coeffs.append_write(1, 2, 3, 4)
coeffs.append_write(1, 2, 3, 4)

```
The amount of RAM you use is O(N) where N is number of arguments you pass in at a time

peterhinch, 2018-08-22:
I appreciate you aim to define a **general** solution. A solution to a **specific** problem like freezing a large array of ints is already possible, along the lines of [font_to_py.py](https://github.com/peterhinch/micropython-font-to-py).

A PC-based utility would take as input the array and output Python code. This would serialise the array into a bytes object and would create a memoryview of that object. It would provide a function which took as input the index into the array and return the int. To do this it would use the memoryview to take the correct 4 byte slice of the bytes object (without copying) and convert it to an int.

The Python output file may then be frozen.

While an _ad hoc_ solution, it's easily done and proven to work: RAM usage when importing a large font file is around 320 bytes.

adritium, 2018-08-22:
What is the step that moves the data from RAM to readonly memory?

adritium, 2018-08-22:
@peterhinch 

I finally understand why your method conserves RAM but in case others do not, I will explain.

Firstly, this method only works if you freeze the `.py` i.e. use the PC utility to go from `.py` to `.mpy` to `.c`.

Say you have this `.py` file: 

```Py
nums = b'\x00\x01\x02\x03'

memv = memoryview(nums)
```

The generated `.c` code will have, among other things, this structure . . .

```C
STATIC const mp_obj_str_t const_obj_frozen_ar__lt_module_gt__0 = {{&mp_type_bytes}, 2949, 4, (const byte*)"\x00\x01\x02\x03"};
```

. . . which is `const` which means it's placed in flash!

The generated bytecode is:
```C
// frozen bytecode for file frozen_array, scope frozen_ar_<module>
STATIC const byte bytecode_data_frozen_ar__lt_module_gt_[33] = {
    0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_frozen_array & 0xff, MP_QSTR_frozen_array >> 8,
    0x46, 0x00, 0x00, 0xff,
    0x17, 0x00, 
    0x24, MP_QSTR_nums & 0xff, MP_QSTR_nums >> 8,
    0x1b, MP_QSTR_memoryview & 0xff, MP_QSTR_memoryview >> 8,
    0x1b, MP_QSTR_nums & 0xff, MP_QSTR_nums >> 8,
    0x64, 0x01, 
    0x24, MP_QSTR_memv & 0xff, MP_QSTR_memv >> 8,
    0x11, 
    0x5b, 
};
```

These 3 lines together load the address of byte array - _which is in flash_ - and then assign it to the Py identifier `nums`

```C
    0x46, 0x00, 0x00, 0xff,
    0x17, 0x00, 
    0x24, MP_QSTR_nums & 0xff, MP_QSTR_nums >> 8,
```

Ta-da! The bytes stay in flash but we have a memoryview so we can access them.

Maybe I have to initialize `memoryview` with **read-only** but other than that, finished.

adritium, 2018-08-22:
@peterhinch 

Your solution works but it requires that you freeze the `.py` which means once the firmware is already built, you have no recourse.

There's still plenty of cases where you cannot freeze but you want to save RAM by moving an object from RAM to _readonly_ memory.

closed 2022-06-23 by dpgeorge