micropython/micropython #4068: Cant use VSPI and HSPI at the same time on ESP32

vstenb, 2018-08-21:
Trying to use two SPI buses fails with the errors "OSError: SPI device already in use" and "E (1352179) spi_master: spi_bus_initialize(152): dma channel already in use" on the ESP32.

The following code produces the error:

from machine import SPI
spi1 = SPI(1)
spi2 = SPI(2)

adritium, 2018-08-21:
Looks like this is a known issue

https://github.com/micropython/micropython/blob/d1fd889ad072afb1e7ffe4c0e5f4ef258226d70f/ports/esp32/machine_hw_spi.c#L188-L200

It's using the same DMA channel for all creation of SPI objects which is not allowed.

https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/spi_master.html#_CPPv218spi_bus_initialize17spi_host_device_tPK16spi_bus_config_ti

> dma_chan: Either channel 1 or 2, or 0 in the case when no DMA is required. Selecting a DMA channel for a SPI bus allows transfers on the bus to have sizes only limited by the amount of internal memory. Selecting no DMA channel (by passing the value 0) limits the amount of bytes transfered to a maximum of 32.

The default `dma_chan` should be zero AND the `SPI` constructor should allow you to select a DMA channel.

vstenb, 2018-08-21:
Thanks for the feedback. I set up the toolchain and compiled with the DMA channel set to zero. It now works with two SPI buses!

mwelling, 2018-08-29:
Also while working on the OHSummit badge I noticed that if you allocate the SPI channel and software reset the board the same error occurs. It seems that the allocation improperly propagates across the system reset.

Here is a gist explain how to use the epaper driver:
https://gist.github.com/mwelling/3381aa45f3dcc4b76c8a7f930106cfb5

Here is what the failure looks like:
MicroPython v1.9.4-179-gce934ae4 on 2018-08-26; ESP32 module with ESP32
Type "help()" for more information.
>>> from machine import Pin
>>> import gxgde0213b1
>>> import font12
>>> import font20
>>>  
>>> reset = Pin(16, Pin.OUT)
>>> dc = Pin(25, Pin.OUT)
>>> busy = Pin(4, Pin.IN)
>>> cs = Pin(5, Pin.OUT)
>>> epd = gxgde0213b1.EPD(reset, dc, busy, cs)
>>> epd.init()
0
>>> fb_size = int(epd.width * epd.height / 8)
>>> fb = bytearray(fb_size)
>>> epd.clear_frame(fb)
>>> epd.set_rotate(gxgde0213b1.ROTATE_90)
>>> epd.display_string_at(fb, 0, 0, "Hello World!", font20, gxgde0213b1.COLORED)
>>> epd.draw_bmp_at(fb, 10, 10, 'happy180.bmp', gxgde0213b1.COLORED)
>>> epd.display_frame(fb)
>>> 
PYB: soft reboot
OSError: [Errno 2] ENOENT
MicroPython v1.9.4-179-gce934ae4 on 2018-08-26; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> from machine import Pin
>>> import gxgde0213b1
>>> import font12
>>> import font20
>>>  
>>> reset = Pin(16, Pin.OUT)
>>> dc = Pin(25, Pin.OUT)
>>> busy = Pin(4, Pin.IN)
>>> cs = Pin(5, Pin.OUT)
>>> epd = gxgde0213b1.EPD(reset, dc, busy, cs)
E (43850) spi_master: spi_bus_initialize(134): host already in use
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gxgde0213b1.py", line 27, in __init__
OSError: SPI device already in use
>>>

adritium, 2018-08-29:
@mwelling I have a very short test for you to run to pinpoint cause of that error.

Before you call `gc_init()`, zero out the heap that you pass in and rerun your test with the soft-reset.

uPy does NOT reset the heap's contents, only the "accounting" blocks.

https://github.com/micropython/micropython/blob/828f771e327b932afc4865dbec53ce567dce45f5/py/gc.c#L140-L147

mwelling, 2018-09-04:
MicroPython v1.9.4-179-gce934ae4-dirty on 2018-09-03; ESP32 module with ESP32
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import Pin
=== import gxgde0213b1
=== import font12
=== import font20
===  
=== reset = Pin(16, Pin.OUT)
=== dc = Pin(25, Pin.OUT)
=== busy = Pin(4, Pin.IN)
=== cs = Pin(5, Pin.OUT)
=== epd = gxgde0213b1.EPD(reset, dc, busy, cs)
=== epd.init()
=== fb_size = int(epd.width * epd.height / 8)
=== fb = bytearray(fb_size)
=== epd.clear_frame(fb)
=== epd.set_rotate(gxgde0213b1.ROTATE_90)
=== epd.display_string_at(fb, 0, 0, "Hello World!", font20, gxgde0213b1.COLORED)
=== epd.draw_bmp_at(fb, 10, 10, 'happy180.bmp', gxgde0213b1.COLORED)
=== epd.display_frame(fb)
=== 
0
>>> 
MicroPython v1.9.4-179-gce934ae4-dirty on 2018-09-03; ESP32 module with ESP32
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import Pin
=== import gxgde0213b1
=== import font12
=== import font20
===  
=== reset = Pin(16, Pin.OUT)
=== dc = Pin(25, Pin.OUT)
=== busy = Pin(4, Pin.IN)
=== cs = Pin(5, Pin.OUT)
=== epd = gxgde0213b1.EPD(reset, dc, busy, cs)
=== epd.init()
=== fb_size = int(epd.width * epd.height / 8)
=== fb = bytearray(fb_size)
=== epd.clear_frame(fb)
=== epd.set_rotate(gxgde0213b1.ROTATE_90)
=== epd.display_string_at(fb, 0, 0, "Hello World!", font20, gxgde0213b1.COLORED)
=== epd.draw_bmp_at(fb, 10, 10, 'happy180.bmp', gxgde0213b1.COLORED)
=== epd.display_frame(fb)
=== 
E (50965) spi_master: spi_bus_initialize(134): host already in use
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "gxgde0213b1.py", line 27, in __init__
OSError: SPI device already in use

mwelling, 2018-09-04:
Here is the change that I made:
```
diff --git a/ports/esp32/main.c b/ports/esp32/main.c
index acbbfdcc..cfa75383 100644
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -77,6 +77,9 @@ void mp_task(void *pvParameter) {
     void *mp_task_heap = malloc(mp_task_heap_size);
 
 soft_reset:
+    // zero out heap
+    memset(mp_task_heap, 0,
+          mp_task_heap_size);
     // initialise the stack pointer for the main thread
     mp_stack_set_top((void *)sp);
     mp_stack_set_limit(MP_TASK_STACK_SIZE - 1024);
```

adritium, 2018-09-04:
Does that help?

Elsewhere, I was reminded that when allocating the new block is zeroed if a flag is set.

adritium, 2018-09-05:
Let's move this question to another issue since the initial question has been resolved #4103

mwelling, 2018-09-07:
Okay sorry for the noise on this issue.

dpgeorge, 2019-01-23:
Should be fixed by d82f344f61811687faee67f4b8d3b4bd333e9f32

closed 2019-01-23 by dpgeorge