micropython/micropython #4069: Unable to build from master branch

kamikaze, 2018-08-21:
```
...
CC ../../lib/lwip/src/core/ipv6/nd6.c
CC ../../lib/lwip/src/netif/ethernet.c
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: error: build-PYBV11/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/7.3.0/libgcc.a(_udivmoddi4.o) does not
arm-none-eabi-ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/7.3.0/libgcc.a(_udivmoddi4.o)
make: *** [Makefile:522: build-PYBV11/firmware.elf] Error 1
```

My GCC:
```
$ LC_ALL=C arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/7.3.0/lto-wrapper
Target: arm-none-eabi
Configured with: /var/tmp/portage/cross-arm-none-eabi/gcc-7.3.0-r1/work/gcc-7.3.0/configure --host=x86_64-pc-linux-gnu --target=arm-none-eabi --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/arm-none-eabi/gcc-bin/7.3.0 --includedir=/usr/lib/gcc/arm-none-eabi/7.3.0/include --datadir=/usr/share/gcc-data/arm-none-eabi/7.3.0 --mandir=/usr/share/gcc-data/arm-none-eabi/7.3.0/man --infodir=/usr/share/gcc-data/arm-none-eabi/7.3.0/info --with-gxx-include-dir=/usr/lib/gcc/arm-none-eabi/7.3.0/include/g++-v7 --with-python-dir=/share/gcc-data/arm-none-eabi/7.3.0/python --enable-languages=c,c++ --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 7.3.0-r1 p1.1' --disable-esp --enable-libstdcxx-time --enable-poison-system-directories --with-sysroot=/usr/arm-none-eabi --disable-bootstrap --with-newlib --enable-multilib --disable-altivec --disable-fixed-point --disable-libgcj --disable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --disable-libmpx --disable-vtable-verify --disable-libvtv --disable-libquadmath --enable-lto --without-isl --disable-libsanitizer --enable-default-pie --enable-default-ssp --disable-libstdc++-v3
Thread model: single
gcc version 7.3.0 (Gentoo 7.3.0-r1 p1.1)
```

adritium, 2018-08-21:
This is not a forum.

This is the forum https://forum.micropython.org/index.php

stinos, 2018-08-22:
Well, perhaps this is a bug report, so it has it's place: stm32 port should build with arm-none-eabi-gcc but it seems it doesn't in this case.

@kamikaze  Would help to show the build command line though, and specify whether/when it used to work previously.

dhylands, 2018-08-22:
It builds fine for me (ubuntu 18.04.1). I'm using the arm compiler from here: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
```
> arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```
I also upgraded the toolchain to gcc-arm-none-eabi-7-2018-q2-update:
```
> arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
```
and that worked as well. I'd recommend trying the toolchain from the URL above.

kamikaze, 2018-08-22:
That's my command, @stinos

`alias makemp="(cd ~/projects/micropython/mpy-cross/ && make clean && make && cd ../ports/stm32/ && make clean BOARD=PYBV11 && LC_ALL=C make BOARD=PYBV11 CFLAGS_EXTRA=\"-fno-pie\")"`

dhylands, 2018-08-23:
For what its worth, building with 
```
make BOARD=PYBV11 CFLAGS_EXTRA=-fno-pie
```
also works fine for me (using gcc 7.3.1 from arm)

dpm76, 2018-08-31:
I had compiled the port for my STM32-NUCLEO F767ZI successfully in the end of June. That time I've used the tool chain of arm-none-eabi, version 4.9.3, included in the default Ubuntu 16.04 repository. Recently, I've pulled from Github and surprisingly had the same error as @kamikaze describes.
Then I downloaded the tool chain version gcc-arm-none-eabi-7-2018-q2, as @dhylands recomends, and now it compiles again. Just tell to the make-tool the tool chain to use with the CROSS_COMPILE variable.

In deed, this is not a bug, but an action could be taken: to correct the Makefile script to have compatibility with old tool-chains, or to tell the minimal tool-chain version at the compiling instructions.

dpgeorge, 2022-03-30:
Closing due to inactivity (and master builds well on all recent gcc versions, up to v11).

closed 2022-03-30 by dpgeorge