micropython/micropython #4071: add ability to create big int for freezing bytecode

adritium, 2018-08-21:
Is it a limitation of `mpy-cross` that you cannot create non-trivial objects?

```Py
a = 1073741824
```

```C
// frozen bytecode for file frozen_array, scope frozen_ar_<module>
STATIC const byte bytecode_data_frozen_ar__lt_module_gt_[21] = {
    0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_frozen_array & 0xff, MP_QSTR_frozen_array >> 8,
    0x00, 0x00, 0xff,
    0x17, 0x00, 
    0x24, MP_QSTR_a & 0xff, MP_QSTR_a >> 8,
    0x11, 
    0x5b, 
};
```

```Py
a = 1073741823
```

```C
// frozen bytecode for file frozen_array, scope frozen_ar_<module>
STATIC const byte bytecode_data_frozen_ar__lt_module_gt_[25] = {
    0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_frozen_array & 0xff, MP_QSTR_frozen_array >> 8,
    0x00, 0x00, 0xff,
    0x14, 0x83, 0xff, 0xff, 0xff, 0x7f, 
    0x24, MP_QSTR_a & 0xff, MP_QSTR_a >> 8,
    0x11, 
    0x5b, 
};
```

peterhinch, 2018-08-22:
Frozen modules containing long ints work as expected. Please could you clarify your query.

adritium, 2018-08-22:
If I freeze `a = 1073741824` I get `0x17, 0x00,` but freezing `a = 1073741823` gives `0x14, 0x83, 0xff, 0xff, 0xff, 0x7f,`