micropython/micropython #4072: ESP8266 + 4xENC28J60

lwqcz, 2018-08-24:
Hi there,
I know that officialy supported are W5k family ethernet modules, but this type is quite expensive in comparsion to ESP8266 (or ESP32).

Is there any possibility to support ENC28J60 and especialy in multiple boards at once? I mean - I would love to use ESP8266 with one ENC to connect to weatherstation, one ENC to connect to sensor station measuring water level in well, one ENC to connect to another sensor box and finaly last ENC to connect to uplink connectivity. Using wifi would be impractical due to obstacles like thick walls or metal enclosures.

Please let me know if this is possible. Thank you a lot.

PS: If anyone found this feature request interesting, please vote for it by adding 👍  to this post! Thanks a lot!

BTW: I would love support even for one ENC, for more connectivity I will use a switch ...

lwqcz, 2018-09-18:
Please ... Any update or even comment on this? @dpgeorge

drawkula, 2018-09-18:
Probably you'll get more feedback in an ESP related forum.
Or at Hackaday, e.g. in [Ethernet Controller Discovered in the ESP8266](https://hackaday.com/2016/04/01/ethernet-controller-discovered-in-the-esp8266/).

lwqcz, 2018-09-20:
Thx @drawkula but I want micropython implementation, no Arduino sketch rubbish ...

drawkula, 2018-09-20:
> Arduino sketch rubbish

[BS! Its not Arduino based.
You didn't even look at the code.](https://github.com/cnlohr/espthernet/blob/gh-pages/Makefile#L23-L27)

I'll plug out of this issue/thread. Bye!

dpgeorge, 2018-09-21:
> Is there any possibility to support ENC28J60

Yes it would be possible.  This chip provides PHY+MAC via a SPI interface.  It would hook into lwIP to provide a seamless Ethernet interface.  Multiple ENC's would be possible, as long as there is an extra 2 pins for each additional ENC (on top of the 3 SPI pins).

lwqcz, 2018-09-26:
Sorry @drawkula I do not mean any offense or so - I just mean that I want micropython implementation and yes I have look for numerous Arduino sketch based projects, but still I do not want to have Arduino environment on ESP.

@dpgeorge - thank you a lot for your text. It really encouraged me to dive into W5x implementation as inspiration for ENC one - and write some stuff 😀 I understand that ENC support is not something interesting to you to deal with it as there are another tasks/issues or so. Thank you again.

closed 2020-11-29 by lwqcz