micropython/micropython #4073: Save RAM: emit code directly to flash/filesystem/readonly-memory

adritium, 2018-08-24:
Saving RAM is a constant theme for people using micropython.

A super-simple action a user can take to save RAM is to use frozen bytecode but that's only possible if the `.py` is available to be compiled in with the firmware.

If that's not an option, the user can still save the `.py` to flash to at least not occupy RAM and simply pass the pointer-to-flash to `mp_lexer_t *mp_lexer_new_from_str_len()`

But this still leaves a huge source of (unnecessary) RAM consumption during the compilation process and the eventually output.

SUGGESTION: the memory used during the compilation process that is O(1), with respect to the size of the input, should remain in RAM but the O(N) stuff should be optionally placed in flash/nonvolatile memory.

peterhinch, 2018-08-25:
What is the use case? Is it that of an application where large scripts are passed to it at runtime for compilation and running?

In normal development I've yet to encounter a problem not solved by splitting the code into smaller modules or by cross-compilation - the latter with or without freezing.

While I have little knowledge of the compilation phase I'd be concerned about flash write speed and wearout.

adritium, 2018-08-26:
That was just an example. Even if you don’t have a large amount of coefficients but a large amount of code, the bytecode could still consume lots of RAM. 

I know you could split up the file but that requires the third party to do it. If you have an application that accepts random `.py` to execute and you say “Error: file too big, heap overflow by 3728 bytes” … what is the user supposed to do with that?

Yes you could also run out of flash if you implement this “save artifacts to flash” but at that point you would have done all you possibly could.

gpshead, 2019-01-27:
From a usability perspective, getting rid of the need for users to know about and run `mpy-cross` on their .py files and that they should manually compile and place .mpy files on device instead of just saving .py files directly to the flash on the device would be a nice win.

Effectively this would be micropython automatically mpy-cross'ing .py files on device (assuming space was available). Such a feature opens the same cans of worms that .pyc files have had with CPython: guaranteeing that the .py and .mpy file are in sync. But if you do that with a hash (similar to what we implemented in https://www.python.org/dev/peps/pep-0552/), it could still work. With an impact on startup time to verify the hash when both .py and .mpy are both present.

A downside? Such a feature is more interesting on smaller the devices (the opposite of where microcontrollers are headed: plentiful flash and ram) but would increase micropython code size a bit (it's already a tight squeeze on many M0s)

peterhinch, 2019-01-27:
> micropython automatically mpy-cross'ing .py files on device

A major use for mpy-cross is where the target device has insufficient RAM to compile the file (but has enough to run it). In that instance the cross-compilation must be done on the PC.

There is also merit in storing at least the smaller files (main.py and boot.py in particular) as .py files: it is very convenient to use **rshell** to edit these in-place.

dpgeorge, 2022-03-30:
Static bytecode/.mpy files was implemented in f2040bfc7ee033e48acef9f289790f3b4e6b74e5, which will eventually help to implement something similar to what's discussed in this issue.

closed 2022-03-30 by dpgeorge