micropython/micropython #4074: framebuf module: clarify behavior of .line() methods

pfalcon, 2018-08-25:
Description of .line() http://docs.micropython.org/en/latest/pyboard/library/framebuf.html?highlight=framebuf#framebuf.FrameBuffer.line says:

> The line method draws the line up to a second set of coordinates whereas the hline and vline methods draw horizontal and vertical lines respectively up to a given length.

It's not exactly clear whether .line() draws up to and including x2, y2, and excluding it. Knowing this may be useful when drawing many line segments, next starting where previous ends (to avoid potential visual artifacts).

pfalcon, 2018-08-25:
For comparison, SDL2's SDL_RenderDrawLine() explicitly state that both points are included https://wiki.libsdl.org/SDL_RenderDrawLine :

> SDL_RenderDrawLine() draws the line to include both end points. If you want to draw multiple, connecting lines use SDL_RenderDrawLines() instead.

peterhinch, 2018-08-25:
A test suggests that .line() is inclusive of start and end coordinates so I guess the doc needs clarifying.