micropython/micropython #4075: Support python code in uos.dupterm()

lerouxb, 2018-08-25:
I'm trying to plug some python code into uos.dupterm(), but it looks like a python object that supports readinto() and write() doesn't make the cut. Looks like you can't subclass a base stream to make a stream-like object in micropython either.

Any suggestions for how to go about this? Will I have to resort to writing c?

lerouxb, 2018-08-25:
OK. So I discovered IOBase. I subclassed that. I'm only supporting the write method so far, so my read method returns None as that seems to be the right thing to do if you don't have any input (at the moment). But then that totally breaks the main keyboard input in the unix port's REPL - it just hangs. I've done some experimenting and if I return something it does get output, but keyboard input is still being ignored.

What am I doing wrong? I'm trying to mirror the REPL elsewhere. Output only for now, input later. Testing my code in the unix port before running on a microcontroller.

pfalcon, 2018-08-25:
http://docs.micropython.org/en/latest/pyboard/library/uos.html?highlight=dupterm#uos.dupterm says "Duplicate or switch" for a reason. On unix port, it "switches". Aka, it doesn't work the same way as on bare-metal ports.

lerouxb, 2018-08-25:
If it switches and it returns the previous one, why does it return None?

The documentation also says you have to implement readinto whereas it calls read.

lerouxb, 2018-08-25:
OH I can add `#define MICROPY_PY_OS_DUPTERM       (2)` to `ports/unix/mpconfigport.h`, rebuild and then `dupterm(mystream, 1)`. That seems to work.

closed 2018-08-25 by lerouxb