micropython/micropython #4078: Getting  OSError 261 from machine.Timer on esp32

larryfast, 2018-08-27:
Probably to obscure to worry about but I diagnosed the following problem in the Timer Module on the esp32.

If you mistakenly initialize two Timer objects with the same ID, all timers created after the error fail and report "OSError 261".  Unhelpfully, the Timer objects get created without errors and only fail when you run `mytimer.init(...)`

The code below all works except for the lines followed by error messages.

def mycallback(tmr):
   print(tmr)

from machine import Timer
t1 = Timer(200)
t2 = Timer(201)
t1.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
t2.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
t2 = Timer(201)
t2.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 261

t3 = Timer(205)
t4 = Timer(206)
t1.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
t3.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 261
t2.init(period=2000, mode=Timer.ONE_SHOT, callback=mycallback)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 261

martinnedopil, 2019-01-08:
Same situation happened to me.
For ESP32 is only one timer (-1).
If you are use more then one ESP got stuck.

gregorypruden, 2019-02-11:
I also got OSError: 261 on esp32 and using Machine.Timer.

In my case, It happened when I was troubleshooting and so hitting Ctrl-D often.
I noticed that when I did machine.reset() it no longer happened.
I was using timer 0

I was ending my main.py with a timer so that I could test things on the command line.  

```
timer = Timer(0)
timer.init(period=60000, mode=Timer.PERIODIC, callback=lambda t:cb_refine())
```

after pressing Ctrl-D the OSerror: 261 happens when it reaches this line again for the second time.

> Traceback (most recent call last):
>   File "main.py", line 64, in <module>
> OSError: 261
> MicroPython v1.10 on 2019-01-25; ESP32 module with ESP32

gregorypruden, 2019-02-11:
I just changed it from Timer(0) to  Timer(-1), Timer(1), Timer(2) and Timer(3) and all exhibit the same problem when using soft reset or Ctrl-D.

gregorypruden, 2019-02-12:
Also I used timer.deinit() before Ctrl-D and the issue goes away.

mailbox4ad4, 2019-02-19:
In my test, it is nothing related to timer.deinit().
I can reproduce by following procedure:
1. Hardware Reset
2. Download Run (for example: timer_test.py)
3. reFlush tree Error
4. reconnect UART
5. 			               <---Auto run step2 timer_test.py (I don't know why)
6. Stop
7. Download Run
8. Stop
9. Download Run		<--OSError 261

I suspect that this issue is cause by "STOP" command will not reset system, therefore when I donwnload and run again, it will cause two identical timer ID running on the background, then generate OSError:261.

Is it reasonable ?

PS. IDE is uPyCraft v1.1

dybber, 2019-05-24:
Adding the following timer allocation flags to the call to `timer_isr_register` in ports/esp32/machine_timer.c solves the issue (also explained here: https://github.com/espressif/arduino-esp32/issues/1869 )

```#define TIMER_FLAGS    (ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_SHARED)```

I have tested it works with all timers and will make a pull request.

commit de76f73d (2019-05-31): esp32/machine_timer: Reuse Timer handles, deallocate only on soft-reset.

The patch solves the problem where multiple Timer objects (e.g. multiple
Timer(0) instances) could initialise multiple handles to the same internal
timer.  The list of timers is now maintained not for "active" timers (where
init is called), but for all timers created.  The timers are only removed
from the list of timers on soft-reset (machine_timer_deinit_all).

Fixes #4078.
closed 2019-05-31 by dpgeorge