micropython/micropython #4081: Scoped allocation

adritium, 2018-08-27:
@dpgeorge's comment is from [here](https://github.com/micropython/micropython/issues/3774#issuecomment-387963973):

> I'm thinking: @micropython.temp serves as a context which results in a C-call to switch a malloc function
> 
> Yes, I think this is a pretty decent idea. I had a similar idea for a while which is "scoped allocation", whereby all allocations within a function call are allocated in a new "heap scope", and then the entire scope is freed upon exit of the function. Eg:
> 
> @micropython.heap_on_stack(200)
> def foo(x, y):
>     l = [x, y]
>     return len(l)
> The idea here is that the function foo() is wrapped in a new heap and the semantics of calling foo are:
> 
> 200 bytes are allocated on the C stack and used to create a temporary "heap"
> allocation on this heap is simply a matter of incrementing a pointer and checking that it didn't go beyond the extents (of 200 bytes in this case); there is no GC'ing of this temporary heap
> all malloc/free functions in the runtime are switched to use this temporary heap for the duration of the foo execution
> the function foo runs, allocates a new list, computes its length, and returns this integer
> as foo returns the temporary heap is destroyed, along with all objects created on it (in this case the list l)
> the result of foo is a small int and doesn't need the heap and is returned to the caller
> The execution of foo is highly deterministic, in terms of memory use and execution time. Using scoped allocation allows you to allocate in a hard interrupt. The downside of this approach is that foo must not leak any scope-allocated objects into the outer scope, because these objects won't exist once foo returns. Note also that you can nest the allocation scopes.

adritium, 2018-08-27:
Another relevant comment is [here](https://github.com/micropython/micropython/issues/3792#issuecomment-393032218)

> In my view fragmentation is crucial: more important even than performance.
> 
> Yes, I agree. No point having performance if there's no memory to perform in :) That's why I think fragmentation needs to be solved completely, not just alleviated in certain cases (even alleviated in 99% of cases wouldn't be enough). Of course the crude solution is to just issue a reset periodically, but that's pretty crude. I think scoped allocation as discussed in #3774 (comment) is the way to go.

adritium, 2018-08-27:
I understand the Py-heap on C-stack makes sense for most cases but if it's not too much trouble, I would like the option to allocate the Py-heap within an existing Py-heap.

Why?
To make accounting of uPy's memory consumption simpler when designing an embedded system.

Plus we already have a mode that avoids using C-stack when executing bytecode.

```C
// Avoid using C stack when making Python function calls. C stack still
// may be used if there's no free heap.
#ifndef MICROPY_STACKLESS
```

peterhinch, 2018-08-28:
The downside is that the process is no longer deterministic as the allocation may trigger GC in the outer heap. It could also fail if the outer heap became fragmented. Allocating on the stack is fast, deterministic and bound to succeed provided sufficient RAM is allocated for stack.

Determining worst-case stack size is not easy. But making this change doesn't remove its necessity, or in my view simplify it much, in systems designed to be resilient.

adritium, 2018-08-28:
Unless there’s an implementation reason to not allow the OPTION … why not allow it?

adritium, 2018-08-28:
There should be the option to isolate the enclosing `.py` code by setting `MP_STATE_THREAD(dict_globals) = NULL` to run it completely isolated from the caller for the case of [process isolation](https://forum.micropython.org/viewtopic.php?p=29692#p29695).

adritium, 2018-08-28:
@peterhinch 

> The downside is that the process is no longer deterministic as the allocation may trigger GC in the outer heap.

This is only a concern for interrupts.

>  It could also fail if the outer heap became fragmented. Allocating on the stack is fast, deterministic and bound to succeed provided sufficient RAM is allocated for stack.

Some designs might require smaller stack usage but enable higher heap usage.
Unless there’s an implementation reason to not allow the OPTION … why not allow it?

peterhinch, 2018-08-29:
> This is only a concern for interrupts.

Other situations exist requiring deterministic execution, e.g. interfacing to hardware, performing time-critical measurements.

> Unless there’s an implementation reason to not allow the OPTION … why not allow it?

Fair point.

adritium, 2018-08-29:
I wonder if it makes _logical_ sense to also allow the scoped-heap to be allocated through an OS call to access another source of memory?

This would be useful if (a) your OS environment restricts usage of stack and (b) you have different speeds of RAM like Tightly-Coupled-Memory (TCM).