micropython/micropython #4087: mp_obj_list_append - allocation policy not sensitive to RAM usage

adritium, 2018-08-30:
https://github.com/micropython/micropython/blob/828f771e327b932afc4865dbec53ce567dce45f5/py/objlist.c#L234-L244

When the user appends ONE additional item to the list, the underlying size is _always_ doubled.
On PC, with gigabytes of RAM this is fine but not in our `micro` environment.

We should change it to something other than doubling.

stinos, 2018-08-31:
> On PC, with gigabytes of RAM this is fine

Unless your application wants to allocate gigabytes of RAM :)

> We should change it to something other than doubling.

Not sure if it's that simple. If anything, it should be configurable. But this should then probably be made confgurable at compile time, or we end up with non-standard Python code for modifying list allocation policy since the standard has no notion of that (e.g. having a `reserve()` method sounds nice but makes the code not work in any other Python version)

But even if it can be changed, the question then is, to what? This is non-trivial to solve. There is no allocation policy which is a good fit for all considering both performance, fragmentation and memory usage. In extreme cases there's not even 1 allocation policy which is a good fit for an entire application: the best allocation policy for a particular piece of code is the one which takes that piece of code in consideration so there's an infinite number of optimal allocation policies..

tldr; given that the out of memory problem can be normally be solved using preallocation+indexing instead of append anyway it's probably easier to stick with some widely-used default..

adritium, 2018-08-31:
There is no method to preallocate a list.

adritium, 2018-08-31:
You are correct but that still doesn’t solve the problem but when you append one additional item to a list, the list size doubles if it’s full.

pfalcon, 2018-08-31:
Because division by ten is expensive operation. But it's definitely should be made  *1.5 ASAP. It has to be multiplication (i.e. exponential growth) to ensure 
"asymptotically O(1)" behavior for such a common operation as constructing list item by item.

adritium, 2018-08-31:
We're dealing with severely memory constrained situations where asymptotic complexity cannot be the only, or even primary, consideration.

No matter how small the multiplication factor, you're still going to run into issues that [confuse the user](https://forum.micropython.org/viewtopic.php?f=3&t=5074&start=10#p29755)

dhylands, 2018-08-31:
The Stack Overflow solution seems to be to preallocate the list using `[None] * size` https://stackoverflow.com/questions/311775/python-create-a-list-with-initial-capacity
This is probably the only truly portable solution, and it will give you a list of exactly the size you want. You could create your own wrapper class for managing your array which then expands things however you see fit.

MicroPython could also have some micropython specific list behaviour which allows a user to specify a multiplier (i.e. use 1.3 instead of 2) or an addition (i.e. increase list by 10 elements) and the user can tune things how they like but not have portable code. Or perhaps, this could be specified as a compile time option.

adritium, 2018-08-31:
The usage of the list has nothing to do with how its underlying size is managed: `list.append()` only cares about whether there's additional memory or not.

This has nothing to do with portability of `.py` code.

adritium, 2018-09-02:
@dhylands @stinos @pfalcon

 The actual problem is not that `list` takes “too much space“  but that the user sees that there is abundantly free space but cannot append one more item. 

 Of course it’s not the users business to know how much memory the underlying python data structure uses.  I realize that the problem is more of a perception problem: “I have 1000 bytes left yet I cannot append one more integer to this list”. 

 There is a simple fix to this perception problem:  if reallocation, using whatever policy, fails when growing the list then simply reallocate an amount which ends up saturating the memory.

adritium, 2018-09-02:
@pfalcon

In addition to the perception problem, there still is the problem that for memory constrained systems doubling is too big. 

If it has to be a multiply > 1 AND efficient to implement, the divisor should be a power of 2:

X, N integers

Factor = X / 2^N

For any N, X can be 
2^(N+1) < X < 2^ N

We might even consider having the factor be a function of the size of the array: a smaller array should increase faster than a bigger one. 

And of course we need to make that initial array size * initial factor will be > initial array size.

pfalcon, 2018-09-02:
> No matter how small the multiplication factor, you're still going to run into issues that confuse the user: https://forum.micropython.org/viewtopic.php?f=3&t=5074&start=10#p29755

Confusion is such a thing, that it likes to go in layers and circles. For example, "script" by the link above doesn't use list.append(), so if you'd do something to list.append, that probably wouldn't help and user would stay confused.

Otherwise, you're absolutely right that:

1. Allocation policy should be overridable. It was clear from the beginning that 
"one size" won't fit all. That's why there're "// TODO: use alloc policy" comments are sprinkled in the code. It's just nobody came up with both manageable and general way to do that yet.

2. Default allocation policy/ies should be optimized. Like, ASAP. What you say on how to do that also makes sense, just needs to be adjusted for realistic/efficient implementabillity.

jpochyla, 2018-09-03:
For bytearrays, I currently use this trick:

```
b = bytearray(preallocate)
b[:] = bytes()
```

adritium, 2018-09-03:
@jpochyla what is that supposed to do?

peterhinch, 2018-09-05:
I was puzzled by this so I tried it. As far as I can see it works.
```python
$ upython 
MicroPython v1.9.4-479-g828f771 on 2018-09-01; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> a = bytearray(10)
>>> a
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>> id(a)
139849550460928
>>> a[:] = b'\x01\x02'
>>> a
bytearray(b'\x01\x02')
>>> id(a)
139849550460928
>>> a[:] = b'\x06\x07\x08'
>>> a
bytearray(b'\x06\x07\x08')
>>> id(a)
139849550460928
>>> 
```

jpochyla, 2018-09-05:
Oh, when I wrote it I thought it would keep the backing storage, but it seems I've overlooked this line:

https://github.com/micropython/micropython/blob/master/py/objarray.c#L443

I believe `o->free` needs to be adjusted, just as `o->len`, no?

dpgeorge, 2018-09-05:
I agree that `*2` is too aggressive.

> There is a simple fix to this perception problem: if reallocation, using whatever policy, fails when growing the list then simply reallocate an amount which ends up saturating the memory.

Yes, it's probably a good idea to make both changes: 1) turn down the multiplicative growth (to something that is cheap to compute); 2) use `m_renew_maybe` so that if it fails the renew can be retried with a smaller growth (eg by just 1 item).

Actually, there's a TODO in py/misc.h which says "make a lazy m_renew that can increase by a smaller amount than requested (but by at least 1 more element)".  Such a `m_renew_lazy` would be useful here, and probably in other places too.

pfalcon, 2018-09-12:
> I believe o->free needs to be adjusted, just as o->len, no?

Yes.

pfalcon, 2018-09-12:
> I believe o->free needs to be adjusted, just as o->len, no?

This is now tracked as https://github.com/micropython/micropython/issues/4127

jpochyla, 2018-09-12:
> This is now tracked as #4127

Thank you!

jonnor, 2024-09-29:
The behavior seems to be very different in lastest MicroPython - much more space oriented. I now see in `array_append()`:
```
        size_t add_cnt = 8;
        self->items = m_renew(byte, self->items, item_sz * self->len, item_sz * (self->len + add_cnt));
```
So perhaps we can close this issue?

dpgeorge, 2024-11-20:
The list append is still always doubling the memory use when it doesn't have any space left.  That still needs fixing.

> I now see in array_append():

`array` is ok, it's `list` that needs improving.

projectgus, 2024-11-20:
> The list append is still always doubling the memory use when it doesn't have any space left. That still needs fixing.

Does it need fixing, though? It's six years since this mostly theoretical discussion about it. There's a lot of discussion of workarounds, but I don't see anyone actually saying the current behaviour has caused a problem for them.

projectgus, 2024-11-20:
Also worth noting that the trade-off  of growing less aggressively is more potential for memory fragmentation if you are building a big list and doing a lot of allocations of unique incrementing sizes (pretty much worst-case scenario for fragmentation). As pfalcon noted back in 2018, there's no "one size fits all" behaviour (within the constraint of conforming to the Python API for this.)

dpgeorge, 2024-11-20:
> Does it need fixing, though?

At the very least I think it should allow increasing the size of the list even if there's only a small amount of memory remaining, eg:

> use `m_renew_maybe` so that if it fails the renew can be retried with a smaller growth (eg by just 1 item).