micropython/micropython #4089: Adding scan delegates and connecting to external peripherals to ubluepy port

tw595, 2018-08-31:
Right now, unlike the full version of bluepy, scanning is limited to only showing basic information (address, rssi, and scan data). In the full version of [bluepy](https://github.com/IanHarvey/bluepy/blob/master/bluepy/scanner.py) scanners can be assigned delegates and have a wider range of scanning capabilities (not sure about getting characteristics). Are there plans to add something similar to the ubluepy MicroPython port?

This is being tested with the nRF52840 chip on the PCA10056 board.

glennrub, 2018-09-02:
As far as i read the bluepy documentation the ScanDelegate.handleDiscovery(scanEntry, isNewDev, isNewData) does not provide much more info than what is in the ScanEntry. The ScanEntry would not change between the models. However, what will change the behavior on how discovery is performed and add support for getting notifications on discovery of a new device. 
When it comes to data of the ScanEntry the documentation of bluepy lists up a few more fields that could be doable to implement, some of these are directly coupled with the extra parameters in the handleDiscovery. [bluepy - ScanEntry](https://ianharvey.github.io/bluepy-doc/scanentry.html)

There has been some initial work on this already (not finished) in context of notification delegate, which is similar to the discoveryDelegate:
[ubluepy_delegate.c](https://github.com/glennrub/micropython/blob/master/ports/nrf/modules/ubluepy/ubluepy_delegate.c)
[ubluepy_peripheral.c#L133](https://github.com/glennrub/micropython/blob/master/ports/nrf/modules/ubluepy/ubluepy_peripheral.c#L133)

When it comes to plans... ubluepy might get replaced in the future, with a common BLE micropython module that spans across ports, making a uniform BLE API for micropython. For time being there is no such replacement, and it should always be possible to make the ubluepy on top of such future API. So, if you find changes in ubluepy that you would like to be addressed to make it a bit aligned with bluepy i'm open to that. When it comes to the specific delegate for Scanner or fields that might missing from the ScanEntry it would be great to get it in. Would you like to give it a go?

jonnor, 2024-07-27:
ubluepy has been replaced by the low-level `bluetooth` module and the high-level `aioble` package (in micropython-lib). Proposing to close this issue.

robert-hh, 2024-07-27:
ubluepy is still used by the nrf port.

jonnor, 2024-12-27:
Right you are. There is an issue tracking possible switchover in https://github.com/micropython/micropython/issues/9851