micropython/micropython #4090: Traceback - show the character that's causing the bug

adritium, 2018-08-31:
From [the forum](https://forum.micropython.org/viewtopic.php?f=3&t=5074&start=10#p29763):

Also, is there a way to make errors display the character? I just did a beginner error (using = instead of ==), but MPy doesn't show me exactly where it happened:

[image]

I would like an output similar to this:
[image]

adritium, 2018-08-31:
The relevant function is

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/py/obj.c#L81-L106

dpgeorge, 2018-09-05:
This requires a lot of work to implement, and it would use a lot of memory.  The column number of each token would need to be tracked all the way through the parser and compiler (taking memory), and the entire script would need to remain accessible (either in memory or in a file) to print out the text of the line that failed.

adritium, 2018-09-05:
I created this issue more as a documentation of the lack of this feature rather than an actual request.

Zezombye, 2018-09-08:
I think passing the column along with the line would be fine. Then the port could fetch the text itself (as I am doing).

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge