micropython/micropython #4092: ESP8266 cannot read by putty/ampy after cross-compiled

vvg123, 2018-09-02:
With the previous Closed Issue [[Fixed] ESP8266 cannot read by esptool/ampy after cross-compiled #4091](https://github.com/micropython/micropython/issues/4091) and found out another error with below scenario which the **ESP8266 crashed/no respond**:

Due the minimal/basic Micropython features in [**micropython/micropython**](https://github.com/micropython/micropython) so I have to add-in [**micropython/micropython-lib**](https://github.com/micropython/micropython-lib) using cross-compiling (datetime, umqtt and ect).

1. git the Micropython-lib and select out all wanted feature in `.py `files and copy into `ports/esp8266/modules`.
2. Increase the memory size of the unit `0xC0000` as add-in **Micropython-lib** will increase the memory storage.
_Note: If not, will be receive `irom0_0_segment` error)._ 
```
MEMORY
{
    dport0_0_seg : org = 0x3ff00000, len = 0x10
    dram0_0_seg :  org = 0x3ffe8000, len = 0x14000
    iram1_0_seg :  org = 0x40100000, len = 0x8000
    irom0_0_seg :  org = 0x40209000, len = 0xC0000
}
/* define common sections and symbols */
INCLUDE esp8266_common.ld
```

3. Start making file under esp8266 folder.
4. Success to created `firmware-combined.bin` file
```
LINK build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 712584	   1100	  64448	 778132	  bdf94	build/firmware.elf
Create build/firmware-combined.bin
esptool.py v1.2
('flash    ', 35408)
('padding  ', 1456)
('irom0text', 678320)
('total    ', 715184)
('md5      ', 'b33e95108618a576a0a19d4256b925a0')
```
5. Flashing into ESP8266 unit using esptool and the unit able to boot-up.
```
l▒OSError: [Errno 2] ENOENT

MicroPython v1.9.4-479-g828f771e3-dirty on 2018-09-02; ESP module with ESP8266
Type "help()" for more information.
>>>
```
_Note: Getting `OSError: [Errno 2] ENOENT` message even using firmware from [micropython.org](https://micropython.org/download). Would be appreciate if any one can point out the error, thanks._
### Error:
6. Error message show up when connect using `ampy`.
```
D:\My_Coding\Unit_119D65>ampy -p com7 -b 115200 ls
Traceback (most recent call last):
  File "D:\Python\Python36-32\Scripts\ampy-script.py", line 11, in <module>
    load_entry_point('adafruit-ampy==1.0.3', 'console_scripts', 'ampy')()
  File "d:\python\python36-32\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "d:\python\python36-32\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "d:\python\python36-32\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "d:\python\python36-32\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\python\python36-32\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "d:\python\python36-32\lib\site-packages\ampy\cli.py", line 147, in ls
    for f in board_files.ls(directory):
  File "d:\python\python36-32\lib\site-packages\ampy\files.py", line 103, in ls
    raise ex
  File "d:\python\python36-32\lib\site-packages\ampy\files.py", line 96, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "d:\python\python36-32\lib\site-packages\ampy\pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 6, in <module>\r\nAttributeError: \'module\' object has no attribute \'listdir\'\r\n')
D:\My_Coding\Unit_119D65>
```
7. From here the `putty` can not get any respond from the unit and have to do reset.

### Error Conditions: 
**A.** [**OK**] makiefile  and boot-up with [**micropython/micropython**](https://github.com/micropython/micropython) only.

**B.** [**OK**] makefile and boot-up with [**micropython/micropython**](https://github.com/micropython/micropython) + adjust the memory size (step-2).

**C.**  [**Error**] successful makefile and boot-up (shown up in `putty` and working first boot-up) with [**micropython/micropython**](https://github.com/micropython/micropython) + [**micropython/micropython-lib**](https://github.com/micropython/micropython-lib) + adjust the memory size (step-2) but **no respond** via `putty/ampy` after using `ampy` communicate and have to do reset on the unit.
_Note: **OK** means `putty/ampy` able communicate with the unit._

### #Question:
1. Based on the above of finding the esp8266 unit is unstable if because of the size of firmware (>700kb) although successful makefile?
2. Is there a better/correct way to increase/adjust the memory size (different spec of ESP8266) for cross-compiling method?

robert-hh, 2018-09-02:
The ENOENT error in step 5 ist most likely cause by main.py not being present. That is not critical. The ampy errors are hard to analzye. Did you try rshell? Fro file transfer you can also use the built-in webrepl.

vvg123, 2018-09-02:
Hi Robert, thanks of your reply. I don't know rshell yet as I'm new to Micropython and now trying to do cross-compile for freezing the modules. For the above of error the webrepl failed as well because can't login into unit via putty for settings. 

Do you know how to compile Micropython-lib into the `bin` file using cross-compile? Now I only can do is using `upip.install` for the lib after flashed to avoid the above of error (the `bin` file too large >700kb).

robert-hh, 2018-09-02:
The way you  choose to place selected parts of micropython.lin into the modules subdir is the most effective for runtime use. I remeber once haveing change the loader map to increase memory, but did not use finally. What I did instead is removing modules from the binary which I did not need. One good candidate is berkeley-db, which in my opinion shoudl never have been included. It was voted for in the kickstarter campaign, but I do not know why. You can remove that by setting:
MICROPY_PY_BTREE = 0
in the Makefile. That gives you like 25k extra space for frozen bytecode.

vvg123, 2018-09-02:
Great to know that `MICROPY_PY_BTREE = 0 `and its reduce the file size indeed. I'm using **ESP-07S** which have 4MB of flash and theoretically the memory is enough for large `bin` file size (step-2) but unfortunate having error on it. May be I didn't fine tuning the setting on the whole process. 

For being so far is enough to built the frozen module in the unit. Looking forward of the new knowledge on Micropython.

vvg123, 2018-09-02:
Oh I see, but for the above of error the `bin` file didn't exceed 1MB (nearly 800kb included whole `lib`). I guess need to do tuning in somewhere.

vvg123, 2018-09-02:
Robert, need your help for the `ssd1306.py` which I just notice the driver files didn't compile initially (cannot `import ssd1306`). Is that I need to manual copy the `ssd1306.py` into module folder?

vvg123, 2018-09-02:
Hi Robert, I think I know why the unit are crash as you pointing out the memory issue as according to flash capacity `4096` able to increase the `irom0_0_seg` size/len into `0xC0000` (as step 2) and `0xC0000 = 786432 byte`. Means the `bin` file size shall not large than` 786kb` and hope this is the reason of unit crash.

robert-hh, 2018-09-02:
We should continue the discussion in the forum https://forum.micropython.org/, section ESP8266, where there is a larger active community. This place here is the bug tracker.

vvg123, 2018-09-02:
ok, i closed this issue. Thanks pf your help, Robert.

closed 2018-09-02 by vvg123

robert-hh, 2018-09-02:
b.t.w.: I tried again increasing irom0_0_seg  and it worked fine. I only had to erase flash before loading the new image. That is supported by the Makefile with:
make erase

vvg123, 2018-09-04:
Hi Robert, you means add-in `make erase` into Makefile? Or during Step-3 (which I'm using `make clean; make axtls; make`)? 

By the way, are you add-in `Micropython-lib` into `bin` file together with increasing `irom0_0_seg`? That would be at **C-Condition** which i get error on it. If without `lib` files then would be the **B-Condition** which no error.

robert-hh, 2018-09-04:
The ESP8266 Makefile has a few utility "targets" besides the ones for building the code:
`make clean` cleans out everything left over from previous builds. That is required when buidl options are changed
`make axtls` build just the axtls  lib. But that is not required to be called separately, becaus it is also exected with
`make`. Which builds the the firmware
`make deploy` can be used to flash the firmware on the device. It calls esptooly.py with the standard settings
`make erase` can be used to erase the flash. It also calls esptools.py with the erase_flash option.
The targets which deal with the device have some defaults for PORT and BAUD rate, which can be modified, e.g. by `make PORT=/dev/ttyUSB1 deploy`.

When you increase the ìrom0_0_seg`, the there is more room for python code, "frozen" into the firmware. That happens automagically, when you copy python script, e.g. from micropython-lib, into the modules subdirectory before building. In my experience you should have again a fully functional device, which should work with ampy.

vvg123, 2018-09-04:
Robert, you are right.  Now every time I have to `erase_flash `and `write_flash` using esptool which is more proper way. Many thanks of your help, Robert and making me more clear on the process of cross-compile and found out that:

`esp-open-sdk` able to build larger `bin` file (>800kb) but will make the unit ESP-07s (4MB flash) crash. That's the result (not sure for other higher capacity).