micropython/micropython #4095: bug? during parsing error: exception doesn't contain enough info to cleanup file descriptor

adritium, 2018-09-03:
[From forum](https://forum.micropython.org/viewtopic.php?f=3&t=5074&p=29822#p29819)

`mp_parse()`:
https://github.com/micropython/micropython/blob/3cd2c281d7cf990b3afb78287a58cf4ee3f23ca5/py/parse.c#L1141-L1157

`mp_lexer_free()` is called at the end of `mp_parse()` but if a syntax error occurs during the parsing, `mp_lexer_free()` is not called which means a file descriptor would leak.

The exception only contains the filename which is not enough to cleanup the original `fd`.

The entire `lex` needs to be contained in the exception.

adritium, 2018-09-07:
@dhylands is this a valid issue?

dhylands, 2018-09-08:
Not sure - it sounds legit, but I haven't really looked into it.

jonnor, 2024-09-29:
6 years since last update. Is this a problem in recent MicroPython versions?

dlech, 2024-09-29:
It looks like this was fixed by https://github.com/micropython/micropython/commit/5e122b11eaee64bbece2ace08fc3f994d5f83f2e

jonnor, 2024-09-29:
Excellent, thanks for checking :)

closed 2024-09-29 by jonnor