micropython/micropython #4098: NRF52840 OpenThread Support

CraigZych, 2018-09-04:
Are there any plans to support OpenThread on the NRF52840 chip?
There are 2 different Mesh networking technologies natively built for this chip.

Both the nrf52832 and nrf52840 support Bluetooth5 Mesh but only the 840 supports thread.

Personally thread support would be my first choice and I wish I knew where to begin exposing the api to upython. I've played around with OpenThread on the nrf52840 development boards and was very impressed.

I know pycom got openthread working with micropython through a LORA network but supporting the proper radio would be awesome.

https://www.nordicsemi.com/eng/Products/nRF52840
https://openthread.io/
https://www.nordicsemi.com/eng/Products/nRF5-SDK-for-Thread

glennrub, 2018-09-09:
Hi @CraigZych,
I'm sorry to tell that there are no immediate plans of supporting Mesh or OpenThread at this point. However, It is definitely interesting, and it should be technical possible.

Given that you or someone else would invest time in this, I would suggest to clear out a few things before any integration starts. The first would be to check license terms on the existing libraries/drivers and how easy it would be to add support for OpenThread/Mesh into micropython. It would be good to know how tightly integrated the existing protocols are to the SDK components. Could the protocols be used outside SDK context? Is there a lib.a or something that could be used?

CraigZych, 2018-09-12:
Pycom has ported openthread to their esp32 based device while totally changing it to run over lora while significantly improving range(positive) but dramatically increasing latency(huge negative depending on application)

Considering they were able to do that it should be very feasible to implement it on a thread certified piece of hardware like the nrf52840 using the standard 802.15.4 hardware. 

I unfortunately dont have any experience implementing new c modules.

I've tested the pycom implementation with 4 nodes and its rock solid for the most part and there is full access to the underlying openthread cli incase you need to do something within openthread they havent implemented at the python level.

If someone could help or guide me on the initial c module import I'd be happy to try to write and test the interface between micropython and openthread.

I have 10 nrf52840 devices I can test on to make sure its solid.

glennrub, 2018-09-13:
@CraigZych , i took a quick look at the source code at [github.com/openthread](https://github.com/openthread/openthread) and it looks like license wise it is compatible (BSD-3 + Nordic-5 clause and a few exceptions which probably can be sorted out)

> I've tested the pycom implementation with 4 nodes and its rock solid for the most part and there is full access to the underlying openthread cli incase you need to do something within openthread they havent implemented at the python level.

Does this mean that you have two "REPL" in parallel, one for micropython and one for open-tread cli?
Is the idea to make an equivalent to the API of cli in micropython to provide the same low level configuration functions using micropython instead of the cli. Or am i wrong? 

I have not looked at how it is done in pycom port, but just on top of my head i can see this working in micropython with an introduction of a class network.OpenThread() which can do the low level setup of the node, and usocket which provides a socket interface to the configured node, providing IP connection. 

> If someone could help or guide me on the initial c module import I'd be happy to try to write and test the interface between micropython and open-thread.

I can probably guide a bit initially if you like to. I'm active on IRC if you would like to get a more free forum for discussions, micropython or micropython-nrf.

Craigzyc, 2018-09-15:
What is your ICR nickname?

I tried glennrub but it said no such user.

They don't actually have 2 "REPL" in parallel but you can access the underlying from code. Example:
```
ot = network.OpenThread()
print(ot.cli("command")) ##would execute whatever command on the openthread cli and return the result
```

I think the ideal implementation would be to have the following to be consistent with how WLAN operates. 
```
ot = network.OpenThread(
	panID = "0xbeef",
	txPower = 10,
	autoCommission = True,
	autoStart = True,
	channel = 11
)
ot.start()

ot.active()

ot.isconnected()

ot.ifconfig()

ot.coap(
	method = "POST",
	address = "fdde:ad00:beef:0:dbaa:f1d0:8afb:30dc",
	uri = "test",
	payload = "test",
	confirmable = True)

ipaddr = ot.ipaddr()
linkquality = ot.linkquality()

```

I'm trying to include the prebuilt openthread stack in the build as a start but I'm getting 
`arm-none-eabi-gcc: error: build-pca10056-s140/lib/libopenthread-cli-ftd.a: No such file or directory`
I'm obviously not putting the files in the right spot or including them properly.
I'd like to make it build option just like the bluetooth stack but not sure yet where to start.

glennrub, 2018-09-22:
@Craigzyc , you could try the nick c45713.

Craigzyc, 2018-10-08:
I did try to reach out but didn't get a response. I've never used IRC so I hope I did it right.

Either way I've played around with this quite a bit and think I came up with a better approach.

zephyr now has openthread support built so as long as the zephyr port is considered stable enough for use I think it may be easier to use zephyr and create the necessary bindings to use the openthread module.

I'm having issues building zephyr but hopefully can figure this out once I get that sorted out.

bmeisels, 2018-10-29:
@Craigzyc Have you had any progress?
I have been playing with openthread and would be happy to join you to work on getting support into micropython.

Craigzyc, 2018-10-30:
I've been trying to get the zephyr port to build with openthread support. 

If you clone zephyr and build inside the /net/echo-client directory with this command:`cmake -GNinja -DBOARD=nrf52840_pca10056 -DOVERLAY_CONFIG=overlay-ot.conf ..
` it will build zephyr including openthread support. 

If I copy the overlay-ot.conf file and add the -DOVERLAY_CONFIG parameter from that example to line 105 of the MAKEFILE it attempts to build in openthread but doesn't actually download or include it. 

If you first build the zephyr sample without the overlay then add the overlay without removing the build directory it throws the same errors so I'm assuming its some kind of cache issue but it persists if I clear the cache and delete all build folders.

I haven't had too much time to play with it yet.

bmeisels, 2018-11-02:
I went according to your instructions and managed to get it compile with openthread.
I added a specific config file for the pca10056 based on the config file from the echo-client example combined with the overlay-ot.conf.
I think you needed some configuration that openthread is dependent on (such as CONFIG_FLASH).
As i only have a pca10059 i haven't tested this on a pca10056.
Could you please try to run it on your board?
I would like to get this working on the pca10059 as well but there seems to be an issue with the shell console. I will be looking into this next week.

My changes are at:
https://github.com/bmeisels/micropython/tree/zephyr-nrf52-openthread

CraigZych, 2018-11-02:
Thanks so much for your help. 

Having trouble building it and unfortunately don't have time to debug today as we have visitors coming tonight and a busy work day. 

Could you add the hex file somewhere. Maybe send it over https://www.sharedrop.io/rooms/d03e075c-56d8-4290-abca-1a3cb27a6395

bmeisels, 2018-11-03:
@CraigZych I am not sure if this will work out of the box without some glue.
Here are the compiled files:
https://drive.google.com/open?id=1J3Bz2qtaJbn2URj6m0duPflRWLPRrk6b

Craigzyc, 2018-11-05:
I can confirm this is fully working as I can see the debug messages in the console.

bmeisels, 2018-11-05:
Great. Have you tried using usocket?
Also it seems that openthtread is initialized by default by zephyr. I am wondering how to stop this.
Once we take care of that we can implement a POC of your suggested API.

CraigZych, 2018-11-05:
I should have a night or 2 this week to try it out. I didnt try usocket yet.

Informaticore, 2021-08-28:
Hey - is there a more updated or even included way to run micropython and openthread? The project link above gives me a 404.
OpenThread works already using Zephyr and I have seen the port of Zephyr with Micropython, do they work together already to do Thread? I would be very interested in that, any hint is welcome.

bmeisels, 2021-08-29:
@Informaticore I must have erased this by mistake at some point. 
It shouldn't be hard to bring this back to the point it was at so I guess I can try to have a look at it later this week.

adminy, 2023-04-25:
Too complicated it must be if nobody wants to continue this work 😬

bmeisels, 2023-04-26:
Haven't touched this for a long time. If I were to have another go it would probably be using the Zephyr port.

jonnor, 2025-08-10:
The Zephyr port is starting to take shape now for NRF52840. So yes, that will probably be the way to go.