micropython/micropython #4100: __setattr__ not getting called on esp32

larryfast, 2018-09-04:
__setattr__ doesn't get called when I thought it should?  The same code calls __setattr__ when I run it on python3. Attributes get created but __setattr__() but I wanted to add some code in __setattr__().

I'm running: esp32-20180822-v1.9.4-479-g828f771e3.bin

Here's the example code

```python
class Test():
  def __setattr__(self, name, value):
    print("{} <= {}".format(name,value))

a = Test()
a.b = '123'
a.b
dir(a)
```
with output
a.b = '123'     => no output
a.b    =>  123
dir(a) =>    ['__class__', '__dict__', '__module__', '__qualname__', '__setattr__', 'b']

stinos, 2018-09-05:
MicroPython does implement this (and `__delattr__` as well) but it is optional, and not enabled by default for esp32. You need a build from source where MICROPY_PY_DELATTR_SETATTR is set to 1 in ports/esp32/mpconfigort.h

larryfast, 2018-09-05:
Thank you.  I guess it's time to take the next step and compile my own uPy package.

adritium, 2018-09-05:
@larryfast close this issue if your problem is solved