micropython/micropython #4103: ESP32 - SPI not being reset across soft-reset

adritium, 2018-09-05:
Moved from [here](https://github.com/micropython/micropython/issues/4068#issuecomment-418603439)

adritium, 2018-09-05:
@mwelling 

> No the output above it shows that the error still happens.
> 
> Looks like the deinit function needs to be called on soft reset. Is there a way to do this automatically?

adritium, 2018-09-05:
To do it automatically, we need to add a `__del__` method to the `SPI` class so that whenever the object gets deleted, `__del__()` executes and calls `spi_bus_free`.

"Deletion" during a soft-reset is triggered by this line

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/ports/esp32/main.c#L119

mwelling, 2018-10-03:
Adding the __del__ function did not appear to fix the issue.

The only way I could get the issue to go away was to call the lower level EPD soft reset command.

https://github.com/mwelling/micropython/commit/367620208fbf1c0b1f57ff332f0da924e1dceb7f

adritium, 2018-10-03:
@mwelling Show me the code that you added

mwelling, 2018-10-03:
Arg, I will have to recreate it because it looks like it was on a throwaway branch.

adritium, 2018-10-03:
Someone else has this issue from the forum. 

https://forum.micropython.org/viewtopic.php?t=5337

jedie, 2018-10-03:
Yes it's me :wink:
Is there any more info needed here?

adritium, 2018-10-03:
@jedie 

Modify https://github.com/micropython/micropython/blob/34af10d2ef545ecaf2c1824dac679af39d683e98/extmod/machine_spi.c#L129

by adding `__del__` method that calls `SPI.deinit()`

mwelling, 2018-10-03:
Yeah I did that.

adritium, 2018-10-03:
You have to set the “enable final user” macro to 1.

adritium, 2018-10-03:
Show your code for __del__

mwelling, 2018-10-03:
The finaliser appears to be enabled.

https://github.com/micropython/micropython/blob/master/ports/esp32/mpconfigport.h#L29

Let me go back and recreate the patch.

mwelling, 2018-10-03:
`
diff --git a/extmod/machine_spi.c b/extmod/machine_spi.c
index f0c4896c..cbd60ce8 100644
--- a/extmod/machine_spi.c
+++ b/extmod/machine_spi.c
@@ -129,6 +129,7 @@ MP_DEFINE_CONST_FUN_OBJ_3(mp_machine_spi_write_readinto_obj, mp_machine_spi_writ
 STATIC const mp_rom_map_elem_t machine_spi_locals_dict_table[] = {
     { MP_ROM_QSTR(MP_QSTR_init), MP_ROM_PTR(&machine_spi_init_obj) },
     { MP_ROM_QSTR(MP_QSTR_deinit), MP_ROM_PTR(&machine_spi_deinit_obj) },
+    { MP_ROM_QSTR(MP_QSTR___del__), MP_ROM_PTR(&machine_spi_deinit_obj) },
     { MP_ROM_QSTR(MP_QSTR_read), MP_ROM_PTR(&mp_machine_spi_read_obj) },
     { MP_ROM_QSTR(MP_QSTR_readinto), MP_ROM_PTR(&mp_machine_spi_readinto_obj) },
     { MP_ROM_QSTR(MP_QSTR_write), MP_ROM_PTR(&mp_machine_spi_write_obj) },
`
`PYB: soft reboot
E (14411) spi_master: spi_bus_initialize(134): host already in use
Traceback (most recent call last):
  File "_boot.py", line 3, in <module>
  File "ohsbadge.py", line 28, in <module>
  File "gxgde0213b1.py", line 28, in __init__
OSError: SPI device already in use
OSError: [Errno 1] EPERM
OSError: [Errno 1] EPERM
MicroPython 0.3.0-8-g139d93fc-dirty on 2018-10-03; ESP32 module with ESP32
Type "help()" for more information.
>>>
`

adritium, 2018-10-04:
Does the finaliser get called?
You can test this while in REPL: 

1. create a SPI object `myspi`
2. do stuff with it
3. call `del myspi` on the SPI object
4. make sure you have a breakpoint set in the code for `machine_spi_deinit_obj`
5. call `gc.collect()`

The breakpoint should be hit if you have `MICROPY_ENABLE_FINALISER == 1`

dpgeorge, 2019-01-23:
Should be fixed by da72bb6833ce8e43bc663e3d48596b9209d05464

closed 2019-01-23 by dpgeorge