micropython/micropython #4105: Installing Unix version on Ubuntu 18.08 needs more dependencies

TravisJoe, 2018-09-06:
I had to also install these dependencies in order to build the Unix port on 18.08

sudo apt-get install autoconf
sudo apt-get install libtool

Figured it might be worth noting in the documentation

dpgeorge, 2018-09-07:
Do you know what exactly autoconf and libtool were needed for, was it to build libffi?

TravisJoe, 2018-09-07:
Sorry I should of took some screenshots. I am pretty sure it revolves around libffi. If it is important I can spin up another virtual machine to confirm this.

peterhinch, 2018-09-08:
My Mint 17.3 box has autoconf but not libtool. The Unix port builds OK.

johnthagen, 2018-09-15:
Do you mean Ubuntu 18.04? (Next release after that will be 18.10. Not sure what 18.08 is?)

TravisJoe, 2018-09-17:
You are correct, apparently felt like more like an 8 was needed. Fixed the title.

I was able to build it fine after installing the extra packages. Probably just worth adding to the documentation that those dependences might be there and do not install with the packages mentioned.

jonnor, 2023-07-30:
Ubuntu 18.04 is now end of life. Proposing to close this issue. Those still using the OS can still find these hints via google. Thanks for documenting it here!

mattytrentini, 2023-07-31:
Ubuntu 18.04 technically doesn't go [end of life until April 2028](https://wiki.ubuntu.com/Releases). But it has passed through the end of standard support. 

Point still holds however, this issue can be found if the details are necessary.

closed 2023-07-31 by mattytrentini