micropython/micropython #4107: ESP8266: scan function

ingframin, 2018-09-07:
Is it possible to change the network.scan() function to accept the same parameters as in the Arduino version?
In the end, it is just a matter of not fixing the struct with the parameters for the wifi_station_scan function.
I can take it in charge but I need some help to figure out how the code in the modnetwork.c works.
I cannot find where (scan_done_cb_t)esp_scan_cb is instantiated :(

dpgeorge, 2018-12-06:
> Is it possible to change the network.scan() function to accept the same parameters as in the Arduino version?

It depends what those parameters are.  It is important that all MicroPython ports that have wifi (or network) support have a consistent interface for scanning, so the kinds of parameters than scan can accept should be well chosen.

Which parameters did you want to use?

ingframin, 2019-01-09:
I am working out a solution myself by adding a separate module to my build.
I need access to the low level scan functions of the ESP8266 (chose the channel) and to the wifi_send_pkt_freedom() function. However, they are very specific and not even consistent (for ESP32 they are different for example). That's why I decided not to touch the standard implementation and add a separate module.
If it's interesting for the community, I can make a pull request here when I am done, otherwise it will anyway be available on the repository of my university.

mcauser, 2019-01-09:
For wifi_send_pkt_freedom() see:
https://github.com/micropython/micropython/pull/3075

dpgeorge, 2019-01-31:
Since there is no specific scan() parameter that is suggested to add I'll close this issue for now.

closed 2019-01-31 by dpgeorge

gigawatts, 2020-05-11:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiScan.cpp#L57

dpgeorge, 2020-05-14:
> I am interested in the same parameter options.

Please specify which options you want to see added.   As mentioned above, new options must be made consistent across all ports (esp8266, esp32, stm32).

gigawatts, 2020-05-17:
From https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiScan.cpp#L57

**int16_t WiFiScanClass::scanNetworks**
- bool async, 
- bool show_hidden, 
- bool passive, 
- uint32_t max_ms_per_chan, 
- uint8_t channel

Some background: I am trying to create a "WiFi Geiger Counter" in micropython (on an esp32), inspired by this [esp8266 arduino code](https://github.com/anha1/esp8266-wifi-fake-geiger-counter/blob/master/src/main.cpp).  Being able to pick a single channel, and specifying a max time spent on that channel would allow MUCH faster scans, and thus a faster refresh rate on the RSSI value I am seeking.

The other issue I have just run into, related to this project is that running a scan seems to pause all timers (perhaps that is what the async parameter from the espressif arduino esp32 lib is designed for?). I am using a periodic timer to "tick" the speaker on a period related to the RSSI strength (better RSSI, smaller period, faster ticks), but it freezes during each scan.

Here is my work in progress code https://gist.github.com/gigawatts/712bc9be6414abbeac8e002fad8e4f23

ebad84, 2023-12-30:
hi
i'm interested on making deauth attack WITH micropython, or packet injection with it
is it possible?