micropython/micropython #4110: append file hangs on close after sleepy write loop

rhubarbdog, 2018-09-09:
I have written a data logger. It writes data to the SD card, I mount the sd card manually as i have a SKIPSD file in my flash storage. It loops writing data once every few minutes, press the switch to terminate gracefully unmounting the sd card.  I let the program run for a few loops then press the switch. The lights change indicating the loop is terminating. it then just hangs for a bit, terminates my repl session on my laptop, continues to hang eventually the lights switch off, but my file probably will be corrupted.

First time i run the program when there is no data file it works correctly and writes the data to disk.  It's second time i run the program when the data file already exists the program hangs on the closing of the file. I have swapped the `with open("data_file.txt", "a") as f:` to `f=open("data_file.txt", "a")` in this second program it hangs at `f.close()`

here's a listing
[append.py.txt](https://github.com/micropython/micropython/files/2364160/append.py.txt)

adritium, 2018-09-10:
Are you able to read from the file when it does exist?

Try this and report back: before your normal `with open(...)` if `data_file.txt` already exists, read and copy contents to `data_file2.txt`.

This'll tell us if the reading at least works.

dpgeorge, 2018-09-11:
I can't reproduce this problem.  I tried with a PYBv1.0, MicroPython v1.9.4, with SKIPSD in the flash filesystem.  I loaded append.py into the REPL using ctrl-E paste mode then executed main() at the REPL.  I let it write 4 times to the SD card then pressed USR to exit.  Then I executed main() again, let it write 3 times, then pressed USR.  It exited to the REPL cleanly and I could mount the SD card and read the `data_file.txt` file, and it had 7 entries, as expected.

rhubarbdog, 2018-09-11:
I'm using a PYB-v1.1, but i'm not sure what version of micropython i'm using i have a 1.9.3 and a 1.9.4 in my repository. Can i check the pyboard with dfu programmer? Don't know if it makes a difference i'm copying append.py to pyboard flash as main.py. unmounting my pyboard. Resetting  , then starting REPL on a running program

dpgeorge, 2018-09-11:
Ok, I can confirm the bug with the latest code on a PYBv1.1.  It's difficult to reproduce, needs at least a few successful writes before pressing USR.

rhubarbdog, 2018-09-11:
I've tried the examples @adritium requested. I'm unable to create further bugs with either `with open("data_file.txt", "r") as f:` or `with open("data_file.txt", "w") as f:` when data_file.txt already exists

dpgeorge, 2018-09-11:
It looks like the SD card is going to sleep after some time (about 3 minutes) and this causes subsequent accesses to it to fail.  I found the following test to be able to reliably reproduce the problem:
```python
import time, pyb
sd = pyb.SDCard()
sd.ioctl(1, 0) # init
buf = bytearray(512)
sd.readblocks(0, buf)
time.sleep(3 * 60)
sd.writeblocks(0, buf) # this will fail with a timeout
```
I'm not sure exactly what is wrong.

As a temporary solution, I'd suggest putting a `file.flush()` call after doing file access, and then making sure the delay between writes is not larger than 3 minutes.

dpgeorge, 2018-09-12:
Should be fixed by 9fb1f18cf450216e30d28ccd246a596555b09f87.  See that commit message for further details.

@rhubarbdog thanks for taking the time to raise this issue, it was an important one to fix!

closed 2018-09-12 by dpgeorge