micropython/micropython #4113: MCU_SERIES_F7 def used in asmthumb.c but not set

andrewleech, 2018-09-10:
A few years back the MCU_SERIES_F4 & MCU_SERIES_F7 definitions were added to the stm port: https://github.com/micropython/micropython/pull/1410

They appear to have been since removed and/or largely replaces with stm specific defines.

The `MCU_SERIES_F7 ` define is still checked for in (non stm specific) `micropython/py/asmthumb.c` but it's never set anywhere.

dpgeorge, 2018-09-10:
> They appear to have been since removed and/or largely replaces with stm specific defines.

Yes that's correct.  Replaced with macros like `STM32F7`.

> The MCU_SERIES_F7 define is still checked for in (non stm specific) py/asmthumb.c but it's never set anywhere.

That's a good find!  This check should be replaced by something more ARM/Cortex-generic, so that it is enabled whenever there is a cache that needs flushing.

dpgeorge, 2018-09-15:
Should be fixed by 7e3dd9f8a3ece6349b5a50b5801fd5731b0ffe19

closed 2018-09-15 by dpgeorge

andrewleech, 2018-09-15:
Looks great!