micropython/micropython #4116: OneWire timing issue

rnovacek, 2018-09-10:
I'm trying to use MicroPython on Wemos D1 mini (esp8266) to read temperature from DS18b20 sensor.

Reading values from this sensor is very unreliable for me. I get the temperature like once in 20 attempts to read the temperature. Other reads thrown an `OneWireError` exception during `OneWire.reset(True)` method or no device was found. This is the code I used (from the [MicroPython docs](http://docs.micropython.org/en/v1.9.3/esp8266/esp8266/tutorial/onewire.html)):
```python
import time
import machine
import onewire, ds18x20

# the device is on GPIO12
dat = machine.Pin(12)

# create the onewire object
ds = ds18x20.DS18X20(onewire.OneWire(dat))

# scan for devices on the bus
roms = ds.scan()
print('found devices:', roms)

# loop 10 times and print all temperatures
for i in range(10):
    print('temperatures:', end=' ')
    ds.convert_temp()
    time.sleep_ms(750)
    for rom in roms:
        print(ds.read_temp(rom), end=' ')
    print()
```

I tried flashing Arduino to the same board with same sensor and it worked flawlessly.

I compared source codes for both 1-wire libraries and discovered one difference: 
The second sleep in the reset function is 40us for MicroPython, but 70us for Arduino.

Sources:
Arduino: https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.cpp#L187
MicroPython: https://github.com/micropython/micropython/blob/master/extmod/modonewire.c#L37

So I tried following monkeypatch to use the timings from Arduino:
```python
import onewire
def reset(self, required=False):
    self.pin.value(0)
    time.sleep_us(480)
    self.pin.value(1)
    time.sleep_us(70)
    status = self.pin.value()
    time.sleep_us(420)
    return status

onewire.OneWire.reset = reset
```

With this, I get the correct readings every time. There's also different number for the 3rd sleep, but that doesn't seem to matter.

I found that master should wait 15-60us after reset before reading slave presence signal:
https://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf (page 13) and then it has 60-240us to read the presence (I hope I interpreted the datasheet correctly).

TLDR: I think the second sleep time should be changed from 40us to 70us.

rnovacek, 2018-09-10:
Yeah, I tried with and without the 4.7kOhm resistor, tried powering from both 3.3V and 5V. Maybe I have some wrong DS18b20 (tried two, but from same [shop](https://www.aliexpress.com/item/MK00241-Stainless-steel-waterproof-package-type-DS18b20-temperature-probe-temperature-sensor-18B20-1-metre/32658611153.html) - maybe some clones or whatever).

My point is that MicroPython doesn't adhere to the datasheet (that says 15-60us), I think it means at least 60us to be sure.

Here are a couple more 1-wire implementations:
https://github.com/jabelch/onewire/blob/master/timing.c#L29 - 70us
https://github.com/dsiroky/OneWire/blob/master/onewire.c#L22 - 70us
https://github.com/bigjosh/OneWireNoResistor/blob/ESP8266/OneWire.cpp#L171 - 70us

There seem to be quite big consensus on 70us. Is there any reason why MicroPython uses 40us?

robert-hh, 2018-09-11:
Looking at my devices here, the slave responds within 30 µs and pulls the line down for 120µs. Since the maximum response time of the slave is 60µs, it is indeed more safe to wait a little bit longer, like 70-80 µs.

dpgeorge, 2018-10-17:
Thanks for the report about this, and for tracking down the issue.  The fix to change the delay from 40us to 70us was done in 4904663748a480608797fa5e22285f601bbecfff

I tested it on a PYBv1.0, esp8266 and esp32, and it works for me with the new timings.

closed 2018-10-17 by dpgeorge