micropython/micropython #4117: ESP32 - failed to build on machine_rtc

patvdleer, 2018-09-10:
I've summed up my actions here:

(skipping the ESP8266)

https://gist.github.com/patvdleer/7ba89a650791ae5c4b0f583760be15d1

```
In file included from machine_rtc.c:36:
machine_rtc.c:126:34: error: 'machine_rtc_init_obj' defined but not used [-Werror=unused-const-variable=]
 STATIC MP_DEFINE_CONST_FUN_OBJ_2(machine_rtc_init_obj, machine_rtc_init);
                                  ^~~~~~~~~~~~~~~~~~~~
../../py/obj.h:287:38: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_2'
     const mp_obj_fun_builtin_fixed_t obj_name = \
                                      ^~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build/machine_rtc.o' failed
make: *** [build/machine_rtc.o] Error 1
```

adritium, 2018-09-10:
Seems like it belongs on the forum

patvdleer, 2018-09-11:
But this is the fault code, I haven't made any changes

patvdleer, 2018-09-11:
Checking out `v1.9.4` gives me the same result

```
patrick@patrick-X470:~/git/micropython$ git checkout v1.9.4 

...

HEAD is now at 421b84af9 docs: Bump version to 1.9.4.
patrick@patrick-X470:~/git/micropython$ cd ports/esp32/
patrick@patrick-X470:~/git/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/patrick/git/esp-idf
Current git hash:   30545f4cccec7460634b656d278782dd7151098e
Supported git hash: 3ede9f011b50999b0560683f9419538c066dd09e
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
../../extmod/modussl_mbedtls.c:38:10: fatal error: mbedtls/platform.h: No such file or directory
 #include "mbedtls/platform.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
moduhashlib.c:28:10: fatal error: mbedtls/sha256.h: No such file or directory
 #include "mbedtls/sha256.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
../../py/mkrules.mk:70: recipe for target 'build/genhdr/qstr.i.last' failed
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'

patrick@patrick-X470:~/git/esp-idf$ git checkout 3ede9f011b50999b0560683f9419538c066dd09e
warning: unable to rmdir 'components/asio/asio': Directory not empty
warning: unable to rmdir 'components/expat/expat': Directory not empty
warning: unable to rmdir 'components/mbedtls/mbedtls': Directory not empty
M	components/aws_iot/aws-iot-device-sdk-embedded-C
M	components/bt/lib
M	components/esp32/lib
M	components/esptool_py/esptool
M	components/micro-ecc/micro-ecc
Previous HEAD position was 30545f4c Merge branch 'bugfix/btdm_fix_get_bond_list_error_when_connection_with_no_bond' into 'master'
HEAD is now at 3ede9f01 Merge branch 'feature/openssl_cacert' into 'master'

# back to micropython

patrick@patrick-X470:~/git/micropython/ports/esp32$ make

In file included from machine_rtc.c:36:
machine_rtc.c:126:34: error: 'machine_rtc_init_obj' defined but not used [-Werror=unused-const-variable=]
 STATIC MP_DEFINE_CONST_FUN_OBJ_2(machine_rtc_init_obj, machine_rtc_init);
                                  ^~~~~~~~~~~~~~~~~~~~
../../py/obj.h:280:38: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_2'
     const mp_obj_fun_builtin_fixed_t obj_name = \
                                      ^~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build/machine_rtc.o' failed
make: *** [build/machine_rtc.o] Error 1
```

robert-hh, 2018-09-11:
Please 
- move in your esp-idf directory  /home/patrick/git/esp-idf
- issue the command: `git checkout 3ede9f011b50999b0560683f9419538c066dd09e`
- issue the command: `git submodule update --init --recursive`
- go to the ESP32 directory and issue the command `make clean`
and then try to build the esp32 firmware again.

patvdleer, 2018-09-12:
```
patrick@patrick-X470:~$ cd /home/patrick/git/esp-idf
patrick@patrick-X470:~/git/esp-idf$ git checkout 3ede9f011b50999b0560683f9419538c066dd09e
M	components/aws_iot/aws-iot-device-sdk-embedded-C
M	components/bt/lib
M	components/esp32/lib
M	components/esptool_py/esptool
M	components/micro-ecc/micro-ecc
HEAD is now at 3ede9f01 Merge branch 'feature/openssl_cacert' into 'master'
patrick@patrick-X470:~/git/esp-idf$ git submodule update --init --recursive
Submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C': checked out '8bf852db77c360eebfa4b800754fdb90e29ea43e'
Submodule path 'components/bt/lib': checked out '0575731e349ca54b46cfc084a752790c1ddcdbba'
Submodule path 'components/esp32/lib': checked out '898cc43904003cdf8e19fe6bb8dc50c629d09eff'
Submodule path 'components/esptool_py/esptool': checked out '200ab6e39487bef1df9db12a5be5b0682d80d3c1'
Submodule path 'components/micro-ecc/micro-ecc': checked out '14222e062d77f45321676e813d9525f32a88e8fa'
patrick@patrick-X470:~/git/esp-idf$ cd ../micropython/ports/esp32/
patrick@patrick-X470:~/git/micropython/ports/esp32$ make clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
rm -rf build 
patrick@patrick-X470:~/git/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/bootloader/subproject/main/
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/bootloader_support/src/
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/log/
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/micro-ecc/micro-ecc/
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/soc/esp32/
mkdir -p build/bootloader//home/patrick/git/esp-idf/components/spi_flash/
CC /home/patrick/git/esp-idf/components/bootloader_support/src/bootloader_clock.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/bootloader_random.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/bootloader_sha.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/secure_boot_signatures.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/secure_boot.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/flash_encrypt.c
CC /home/patrick/git/esp-idf/components/bootloader_support/src/flash_partitions.c
CC /home/patrick/git/esp-idf/components/log/log.c
CC /home/patrick/git/esp-idf/components/spi_flash/spi_flash_rom_patch.c
CC /home/patrick/git/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/patrick/git/esp-idf/components/soc/esp32/rtc_time.c
CC /home/patrick/git/esp-idf/components/micro-ecc/micro-ecc/uECC.c
CC /home/patrick/git/esp-idf/components/bootloader/subproject/main/bootloader_start.c
LINK build/bootloader.elf
Create build/bootloader.bin
esptool.py v2.2.1
Create build/partitions.bin
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build//home/patrick/git/esp-idf/components/app_trace/
mkdir -p build//home/patrick/git/esp-idf/components/app_update/
mkdir -p build//home/patrick/git/esp-idf/components/cxx/
mkdir -p build//home/patrick/git/esp-idf/components/driver/
mkdir -p build//home/patrick/git/esp-idf/components/esp32/
mkdir -p build//home/patrick/git/esp-idf/components/esp32/hwcrypto/
mkdir -p build//home/patrick/git/esp-idf/components/ethernet/
mkdir -p build//home/patrick/git/esp-idf/components/ethernet/eth_phy/
mkdir -p build//home/patrick/git/esp-idf/components/expat/library/
mkdir -p build//home/patrick/git/esp-idf/components/expat/port/
mkdir -p build//home/patrick/git/esp-idf/components/freertos/
mkdir -p build//home/patrick/git/esp-idf/components/heap/
mkdir -p build//home/patrick/git/esp-idf/components/json/cJSON/
mkdir -p build//home/patrick/git/esp-idf/components/log/
mkdir -p build//home/patrick/git/esp-idf/components/lwip/api/
mkdir -p build//home/patrick/git/esp-idf/components/lwip/apps/
mkdir -p build//home/patrick/git/esp-idf/components/lwip/apps/sntp/
[... 211 lines cut ...]
CC machine_uart.c
CC modmachine.c
CC modnetwork.c
CC network_lan.c
CC modsocket.c
CC modesp.c
CC esp32_ulp.c
CC modesp32.c
CC moduhashlib.c
CC espneopixel.c
CC machine_hw_spi.c
CC machine_wdt.c
CC mpthreadport.c
CC machine_rtc.c
In file included from machine_rtc.c:36:
machine_rtc.c:126:34: error: 'machine_rtc_init_obj' defined but not used [-Werror=unused-const-variable=]
 STATIC MP_DEFINE_CONST_FUN_OBJ_2(machine_rtc_init_obj, machine_rtc_init);
                                  ^~~~~~~~~~~~~~~~~~~~
../../py/obj.h:280:38: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_2'
     const mp_obj_fun_builtin_fixed_t obj_name = \
                                      ^~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build/machine_rtc.o' failed
make: *** [build/machine_rtc.o] Error 1
```

adritium, 2018-09-14:
It's a bug: `machine_rtc_datetime_obj` is pasted twice so `machine_rtc_init_obj` is not used giving you 

> error: 'machine_rtc_init_obj' defined but not used

https://github.com/micropython/micropython/blob/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d/ports/esp32/machine_rtc.c#L150-L154

patvdleer, 2018-09-16:
I was able to fix this but it simply drops me into the next error...

```
/home/patrick/esp/esp-idf/components/driver/rtc_module.c:99:33: error: initialization of '_lock_t' {aka 'int'} from 'void *' makes integer from pointer without a cast [-Werror=int-conversion]
 static _lock_t adc2_wifi_lock = NULL;
                                 ^~~~
/home/patrick/esp/esp-idf/components/driver/rtc_module.c:104:32: error: initialization of '_lock_t' {aka 'int'} from 'void *' makes integer from pointer without a cast [-Werror=int-conversion]
 static _lock_t adc1_i2s_lock = NULL;
                                ^~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build//home/patrick/esp/esp-idf/components/driver/rtc_module.o' failed
make: *** [build//home/patrick/esp/esp-idf/components/driver/rtc_module.o] Error 1
```

patvdleer, 2018-09-16:
Since I kept running into errors I simply updated the `Makefile` to the following:

`micropython/ports/esp32/Makefile@96`

```
CFLAGS_BASE = -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-error=unused-const-variable -Wno-error=int-conversion -Wno-error=sizeof-pointer-memaccess -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DESP_PLATFORM
CFLAGS = $(CFLAGS_BASE) $(INC) $(INC_ESPCOMP)
CFLAGS += -DIDF_VER=\"$(IDF_VER)\"
CFLAGS += $(CFLAGS_MOD)

# this is what ESPIDF uses for c++ compilation
CXXFLAGS = -std=gnu++11 -fno-exceptions -fno-rtti -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-error=parentheses -Wno-error=sizeof-pointer-memaccess -DESP_PLATFORM $(INC) $(INC_ESPCOMP)
```

This eventually gives me a fail:
```

patvdleer, 2018-09-16:
All warnings/errors suppressed:

```
machine_rtc.c:126:34: warning: 'machine_rtc_init_obj' defined but not used [-Wunused-const-variable=]
 STATIC MP_DEFINE_CONST_FUN_OBJ_2(machine_rtc_init_obj, machine_rtc_init);
                                  ^~~~~~~~~~~~~~~~~~~~
../../py/obj.h:280:38: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_2'
     const mp_obj_fun_builtin_fixed_t obj_name = \
                                      ^~~~~~~~
/home/patrick/esp/esp-idf/components/driver/rtc_module.c:99:33: warning: initialization of '_lock_t' {aka 'int'} from 'void *' makes integer from pointer without a cast [-Wint-conversion]
 static _lock_t adc2_wifi_lock = NULL;
                                 ^~~~
/home/patrick/esp/esp-idf/components/driver/rtc_module.c:104:32: warning: initialization of '_lock_t' {aka 'int'} from 'void *' makes integer from pointer without a cast [-Wint-conversion]
 static _lock_t adc1_i2s_lock = NULL;
                                ^~~~
In file included from /home/patrick/esp/esp-idf/components/newlib/include/stdlib.h:18,
                 from /home/patrick/esp/esp-idf/components/cxx/cxx_guards.cpp:15:
/home/patrick/esp/esp-idf/components/newlib/include/sys/reent.h:405:11: warning: unnecessary parentheses in declaration of '_sig_func' [-Wparentheses]
   void (**(_sig_func))(int);
           ^
In file included from /home/patrick/esp/esp-idf/components/newlib/include/stdio.h:47,
                 from /home/patrick/esp/esp-idf/components/esp32/include/esp_err.h:17,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/include/nvs.h:20,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/src/nvs_page.hpp:17,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp:14:
/home/patrick/esp/esp-idf/components/newlib/include/sys/reent.h:405:11: warning: unnecessary parentheses in declaration of '_sig_func' [-Wparentheses]
   void (**(_sig_func))(int);
           ^
In file included from /home/patrick/esp/esp-idf/components/newlib/include/stdio.h:47,
                 from /home/patrick/esp/esp-idf/components/esp32/include/esp_err.h:17,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/include/nvs.h:20,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/src/nvs_item_hash_list.hpp:18,
                 from /home/patrick/esp/esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp:15:
/home/patrick/esp/esp-idf/components/newlib/include/sys/reent.h:405:11: warning: unnecessary parentheses in declaration of '_sig_func' [-Wparentheses]
   void (**(_sig_func))(int);
           ^
/home/patrick/esp/esp-idf/components/spi_flash/partition.c: In function 'load_partitions':
/home/patrick/esp/esp-idf/components/spi_flash/partition.c:179:66: warning: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Wsizeof-pointer-memaccess]
         strncpy(item->info.label, (const char*) it->label, sizeof(it->label));                     
```

dpgeorge, 2018-09-20:
The warning with `machine_rtc_init_obj` is fixed in 185716514f110560adaa35367aa6886023f29120

The other warnings are within the ESP IDF itself so can't be fixed here.  It looks like you are using a custom toolchain built from source, not the precompiled one from Espressif, so that could be the issue.  I'd suggest trying their provided compiler.

closed 2019-05-24 by dpgeorge