micropython/micropython #4119: ESP32   SHA1 on uhashlib is not correct at 2nd or more times.

CUEICHI, 2018-09-11:
I tried this, and result is not same at 1st time and 2nd or more.
is it a bug? 
>>> os.uname() 
(sysname='esp32', nodename='esp32', release='1.9.4', version='v1.9.4-498-g5cd2c7f2e on 2018-09-09', machine='ESP32 module with ESP32')
>>> import binascii, hashlib
>>> hash = hashlib.sha1(b'mango')
>>> binascii.hexlify(hash.digest()) 
b'934aae49f648ed870c9c421829f4cece6643cf86'
>>> binascii.hexlify(hash.digest()) 
b'0ffd8d43b4e33c7c53461bd10f27a5461050d90d'
>>> binascii.hexlify(hash.digest()) 
b'0ffd8d43b4e33c7c53461bd10f27a5461050d90d'
>>> 

>>> os.uname()
posix.uname_result(sysname='Darwin', nodename='rydeen.local', release='17.7.0', version='Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64', machine='x86_64')
>>> import binascii, hashlib
>>> hash = hashlib.sha1(b'mango')
>>> binascii.hexlify(hash.digest()) 
b'934aae49f648ed870c9c421829f4cece6643cf86'
>>> binascii.hexlify(hash.digest()) 
b'934aae49f648ed870c9c421829f4cece6643cf86'
>>> binascii.hexlify(hash.digest()) 
b'934aae49f648ed870c9c421829f4cece6643cf86'
>>>

Jongy, 2019-02-10:
I can reproduce on a0d60c574a96bb873b0e6822d009031fd5fa3a77.

It happens because `sha*_digest` functions call `_final` multiple times. In the SHA1 case, the `mbedtls_sha1_context` is `memset` to 0s, so after 2 `digest()` calls you just lock down on a single hash value. In the SHA256 case it is not `memset` so you keep getting random garbage.

dpgeorge, 2021-05-25:
See #7316 for a change which raises an exception if update/digest is called after a digest is made the first time.  I think that's a simple solution to this problem.

agatti, 2026-05-24:
This was fixed by 6a127810c03d7666f1866f5e48df3c102805dbe8, closing as completed.

closed 2026-05-24 by agatti

dpgeorge, 2026-05-25:
Thanks @agatti for following this up and closing.