micropython/micropython #4126: machine.RTC missing from stm32 on v1.9.4

ryannathans, 2018-09-12:
The docs seem to suggest that there should be an RTC class in the machine module for the stm32 in v1.9.4

http://docs.micropython.org/en/latest/pyboard/library/machine.RTC.html

```
MicroPython v1.9.4-22-ga8467510d-dirty on 2018-09-11; CustomPCB with STM32F439
Type "help()" for more information.
>>> import machine
>>> dir(machine)
['__class__', '__name__', 'sleep', 'DEEPSLEEP_RESET', 'HARD_RESET', 'I2C', 'PWRON_RESET', 'Pin', 'SOFT_RESET', 'SPI', 'Signal', 'UART', 'WDT', 'WDT_RESET', 'bootloader', 'deepsleep', 'disable_irq', 'enable_irq', 'freq', 'idle', 'info', 'mem16', 'mem32', 'mem8', 'reset', 'reset_cause', 'rng', 'soft_reset', 'time_pulse_us', 'unique_id']

```

adritium, 2018-09-14:
It used to be there

https://github.com/micropython/micropython/blob/abb536da499498b94d5c3dc7379fd5136d273f2d/ports/stm32/modmachine.c#L667-L670

I guess they're in the process of moving it to `pyb.RTC`

https://github.com/micropython/micropython/blob/abb536da499498b94d5c3dc7379fd5136d273f2d/ports/stm32/modpyb.c#L176-L178

andrewleech, 2018-10-07:
I just ran into this myself. 

I actually think it hasn't been implemented in machine module yet for stm32, it looks like that `#if 0` section is a placeholder for when it is implemented correctly - it was there in the original copy of `modmachine.c` for stm32: https://github.com/micropython/micropython/blob/504420c51d91da13e6fd42514475049ab5b15c6d/stmhal/modmachine.c

The syntax for `pyb.RTC` is slightly different, it's probably waiting for someone to take on the conversion required to map it to the `machine` interface.

dpgeorge, 2019-07-04:
The `machine.RTC` class for stm32 was enabled in f88cb8a51412d8570e473045adf09699664f553a

Making all the ports' RTC classes consistent is a WIP (and a separate issue to this one).

closed 2019-07-04 by dpgeorge