micropython/micropython #4132: Floating-point behavior is not consistent with IEEE754

Ecco, 2018-09-13:
For some reason, even the most simple floating-point operations yields results that don't match IEEE754. For instance, in uPy, `0.1+0.2 == 0.3` returns `True`, while it is `False` in CPython and every other programming language that uses IEEE754.

One might argue that uPy's behavior is better. In that case it might, but uPy also does some rounding errors, just not the same as everyone else… For example, `1000.4-1000` will yield `0.4000244` in uPy but `0.39999999999997726` in CPython, JavaScript (Chrome, Firefox), Ruby, etc…

Also it looks like uPy's behavior is not matching the Python standard, since https://docs.python.org/3/tutorial/floatingpoint.html explicitly says that `0.1+0.2 == 0.3` should return `False`.

This bug is reproducible on the [uPy live](http://micropython.org/live/) web page.

EmilieNumworks, 2018-09-13:
The bug seems to be in `mp_parse_num_decimal` which parses `1000.4` as `1000.4000000000001`.

pfalcon, 2018-09-13:
What platform do you use? MicroPython leaves floating-point handling to the platform (which means your FPU if you have that, or somebody's software floating-point library if you don't).

To get ahead with this, you would need to provide binary (i.e. hexadecimal) values of the FP numbers you get, that's the only way to assess correctness/incorrectness of it. (Well, approach being able to assess, the actual assessment is a work on its own.)

EmilieNumworks, 2018-09-13:
We've tested that behavior on different platforms (ARM v7m, x86_64), so this bug is not platform-dependent. The bug really is early in the process, at the parsing level: `1000.4` is improperly parsed as `0x408f433333333334` instead of `0x408F433333333333`.

pfalcon, 2018-09-13:
> To get ahead with this, you would need to provide binary (i.e. hexadecimal) 

To be more specific, we should implement float.hex() (guarded by config option, default off - which are the usual rules) to be able to debug there comfortably.

pfalcon, 2018-09-13:
> 1000.4 is improperly parsed as 0x408f433333333334 instead of 0x408F433333333333.

@EmilieNumworks , good research, thanks. I'm sure @dpgeorge will look into this as his time permits.

As you can see, mp_parse_num_decimal() does iterative float multiplications, a recipe for rounding error. I don't know how other folks do it, perhaps keeping to integer value for as long as possible? Worth research too.

Ecco, 2018-09-13:
At least the line `dec_val *= MICROPY_FLOAT_C_FUN(pow)(10, exp_val);` yields a rounding error. For example, when parsing `1000.4`, it tries to do `dec_val = 10004.0 * 0.1` which yields an incorrect result because of a rounding error. Indeed, in IEEE754 world, `1000.4 != (10004.0*0.1)`.

Ecco, 2018-09-13:
See #4133 for a possible fix. Uses the same logic as some implementations of [strtod(3)](https://github.com/enthought/tcl/blob/607dc191fb900621b97f5a8ca3f363b4241f8cb5/compat/strtod.c#L239).

dpgeorge, 2018-12-04:
The behaviour of float parsing was improved in b768cc6ca8ed6d7430b86dc7ccb9ee2391b3a251

closed 2018-12-04 by dpgeorge