micropython/micropython #4135: Idea to aid debugging of GC issues

adritium, 2018-09-13:
This is in reference to [this forum issue](https://forum.micropython.org/viewtopic.php?f=3&t=5245&sd=d#p30112)

> I tried that (enabling verbose mode in mpportconfig I assume you mean) .
That is when i saw that if i moved the stack onto the heap there were only +- 5 sweep (=free) messages (and it kept working) versus +-20 sweeps (and it crashed after suspend/resuming) if it was still in static memory during the first call of gc.collect() (if i call it every 100ms). 

> So that triggered my whole 'the contents of the static stack seem to be ignored' idea and when i looked at the code i thought i found the explanation ('if you are a hammer everything looks like a nail' - wise ).
> 
> So I will have to look again into why these sweep results are so different between the 2 scenarios, but it is a good starting point. And in the meantime I've learned a lot more about the GC :) . And optimized the external memory parameters so that it is now much faster which was needed with my initial stack located on the heap in external sram:).

When the `gc` is freeing things it maybe shouldn't be freeing, it would be amazing to know WHICH things it is freeing.

Ideally, there'd be a `mpconfigport.h` flag to enable storing a string with every allocated block. When this block is freed, a debug message would print "Variable <XYZ> is freed".

But this requires too big change on the user-side as they'd have to modify ALL their allocations with their custom string but there's a simpler way that is more scalable.

adritium, 2018-09-13:
In `misc.h`, all the memory allocation macros are located.

https://github.com/micropython/micropython/blob/4f3d9429b54ccc2d36123c861cd916b1ee15c640/py/misc.h#L60-L66

Using a flag in `mpconfig.h`/`mpconfigport.h`, these macros would be modified to contain the macros `__line__`, `__func__`, `__file__`

```C
#define m_new(type, num, __file__, __func__, __line__) ((type*)(m_malloc(sizeof(type) * (num))), __file__, __func__, __line__ )
```

In `gc_alloc`, a new entry would be added to some table-like structure with entries

1. head block
2. pointer address
3. `__file__`
4. `__func__`
5. `__line__`
6. deleted?

Whenever a free is performed, the entry in table is marked as deleted, instead of immediately deleted from table, so user has chance to see what happened.

Maybe add a callback or iterator to return lines in table modified.

dpgeorge, 2022-06-23:
Closing due to inactivity.