micropython/micropython #4136: python 3 compat : generators have no .__name__

pmp-p, 2018-09-14:
Hi while trying to pause/patch/resume coroutines indexed by their names in a live program i found a little divergence 

```
async def func():pass
print( func.__name__)
```
µpython : `AttributeError: 'generator' object has no attribute '__name__'`

cpython 3.7 : `func`

dpgeorge, 2018-09-14:
I think this was implemented in e2e22e3d7e29985579b2c91c639c71229422f349.  Please try the latest version of MicroPython.

pmp-p, 2018-09-14:
yeah found that both MICROPY_PY_BUILTINS_COMPILE and MICROPY_PY_FUNCTION_ATTRS are unset. So maybe activating those features is not obvious if you did not read an mpconfigport.h for a port you do not care for.

dpgeorge, 2018-09-14:
Yes you're right, this feature is not enabled on esp8266.  So I guess this issue is about whether or not to enable it on esp8266.

jonnor, 2024-09-29:
Proposing to not enable this on ESP8266 by default, since it is severely constrained on program space. Those that need it can use a custom board definition.

dpgeorge, 2024-09-30:
> Proposing to not enable this on ESP8266 by default, since it is severely constrained on program space.

Yes, I agree.

And we now use ROM level features, so it's a bit easier to see what is enabled/disabled on esp8266.

closed 2024-09-30 by dpgeorge