micropython/micropython #4137: SyntaxError when expanding *args?

martijnthe, 2018-09-14:
I'm having trouble expanding `*args`, see repro case below.
I'm using a Pycom LoPy, which I know doesn't include the latest and greatest micropython, so perhaps the issue has been fixed already in the micropython upstream.

```
Pycom MicroPython 1.18.1.r1 [f5d0c68] on 2018-09-06; LoPy with ESP32
Type "help()" for more information.
>>>
>>>
>>>
>>>
>>>
>>> def x(*args):
...     y = [1, 2, *args]
...     print(y)
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in x
SyntaxError: *x must be assignment target
```

robert-hh, 2018-09-14:
Something like that works
```
def x(*args):
    y = [1, 2] + list(args)
    print (y)
```
Besides that, this repository belongs to micropython.org, but your example was made with the Pycom variant. But for this topic, both variants behave the same.

dpgeorge, 2018-09-14:
This is a relatively new feature introduced in Python 3.5 (PEP448), and MicroPython doesn't yet support it.  See issue #1329.

dpgeorge, 2018-09-20:
Since this feature is tracked by the above-mentioned issue, I'll close this one.

closed 2018-09-20 by dpgeorge