micropython/micropython #4140: Introduce abstraction (like in pointer indirection) for memory read/write?

adritium, 2018-09-14:
Memory read/write abstraction - pointer indirection - has been mentioned as a possible route #1168 to implement precise garbage collector but I think introducing such a thing has additional uses.

Remember that micropython is _often_ run on bare-metal (cc @peterhinch ) and we're using C to implement it so we don't have the nifty tricks of operator overloading of C++ (sigh) i.e. we cannot redefine the `*` dereferencing operator.

# Executing from flash
In #4124 I inquire of the possibility of running a `.mpy` from flash and @dpgeorge answers with a straightforward solution. However there is a large constraint: 

> **The other big issue to solve would be making sure that .mpy's live in a location that is 1) memory mapped to the CPU; 2) contiguous.** 

If we had indirection, the code wouldn't have to be contiguous because the VM code's reading of the opcodes - `switch (*ip++)` - would be replaced with `switch(GET(ip))` (or something).

Once you can execute from ANY flash location, after you `import` you save the `mp_raw_code_t` to flash.

# RAM! (my precious)
Everyone wants to save RAM by any tricks in `.py` code but parsing and compilation are a huge low-ish hanging fruit. We could save the parse-tree and other artifacts in flash.

dpgeorge, 2018-09-20:
> If we had indirection, the code wouldn't have to be contiguous because the VM code's reading of the opcodes - `switch (*ip++)` - would be replaced with `switch(GET(ip))` (or something).

This would be a huge overhead for speed (and also code size).

There are really two discussions here: 1) general pointer indirection to help with a compacting collector; 2) supporting custom memory mapping of bytecode so it can remain in external storage.

For the latter, a more efficient and realistic approach would be to have a bytecode cache in RAM, which would hold the most recently used bytecode functions.  Bytecode would be loaded on demand from storage and executed from RAM.

adritium, 2018-09-20:
You ignored `save the parse-tree and other artifacts in flash`

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge