micropython/micropython #4141: Can I hire you ?

Avi-TelnT, 2018-09-15:
Hi
I am looking to hire an ESP32 Python programmer to help me to develop a firework interface for ESP32 I2S for a mic on I2S ICS-43434 IC and play wav using I2S MAX98357A IC. The target is work needed Google speech recognition and Google text to speech from ESP32. 
The development environment must be https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo 
If you have an experience on ESP32 MicroPython_ESP32_psRAM_LoBo Python please reply my email nissim@telnt.com
And call my Skype nissim.test  
The work is milestone-based and payment will be as milestone below,
We will provide the hardware for that :
1.	Expose to LoBO Python the ESP32-IDE /MicroPython/Tools/esp-idf/components/driver/include/driver/I2S.H and I2S.C by integrate them in machine module so dir(machine) will show the I2S interface and commands to set I2S for recording and playing using the ESP32 DMA. 
2.	Use the above machine.I2S to record and play a wav file a digital mic on 
I2S ICS-43434 IC and play wav using MAX98357A IC
3.	Make an audio PTT (Push-To-Talk) instead of PCB button. By detecting the rising level on the recorded audio level, ok voice keyword recognized. 
4.	Doing above but record and play from ram with Flak encoding as needed by Google SR AND TTS. See https://cloud.google.com/speech-to-text/docs/encoding
5.	I need to use SR to get the recognized the Mic audio, as text returns from Google back to ESP32, modify the return text with meaningful information and send to be played by Google TTS.  Like Siri. With ESP32. 
As: 
You Say to Mic I2S ICS-43434 “What is the temperature now?” 
Get back the text in ESP32 Python  “What is the temperature now”
ESP32 Send to TTS “The temperature now is 30 degree” 
The return wav or Flak been played to  using MAX98357A over I2S
6.	So, I believe to expose I2S code from ESP32 C  to Python will not be a big work. 
a.	Google API use wav Audio Encoding name: FLAC Free Lossless Audio Codec 16-bit or 24-bit required for streams. 
b.	So, FLAC can be done in C or Python. But interface must be from Python. See https://goo.gl/mtrwVN 
7.	Maybe this is a good source for FLAC in C https://goo.gl/cCYFTh
Contact :
Nissim Zur
nissim@telnt.com
Skype: nissim.test

peterhinch, 2018-09-15:
This would get a bigger audience in [the forum](https://forum.micropython.org/). This is the MicroPython bug tracker.

nickzoic, 2018-09-17:
Yep.  Nissim, please post something in the forum.  This is the bug tracker for the mainstream micropython, and so LoBoris fork stuff is out of scope, plus it isn't quite the right place for the level of detail you're going into.  But I think the I2S module would be quite generic and able to be created in mainstream micropython and then merged into your LoBoris fork.

There's been a little previous discussion at https://github.com/micropython/micropython-esp32/issues/185 but I never got around to copying that over to the main repo when it got merged.  I had intended to do some work on this and have some hardware on hand, but have not got around to it.

However, I do appreciate that you're offering to pay for development rather than just expecting it happen by magic, and I hope someone here has a chance to take you up on it.

closed 2018-09-20 by dpgeorge