micropython/micropython #4145: utime.localtime() returns UTC and local time in some cases

lumbric, 2018-09-15:
In Python [time.localtime()](https://docs.python.org/3/library/time.html#time.localtime) returns local time, but utime.localtime() usually returns a UTC time stamp, only in the unix port it returns local time. I guess the correct way should be local time of course, but not having a consistent way localtime() behaves between ports seems to be even worse. I don't have a good suggestion how to fix it, except implementing #2130 (and breaking backward compatiblity?). But maybe unix port can be improved as a first step?

(If you consider this ticket part/duplicate of #2130, please close as won't fix.)

Also related: #3087.

dhylands, 2018-09-16:
It only returns a UTC timestamp if you set it to a UTC timestamp. If you set the time to be localtime then you get back localtime.

lumbric, 2018-09-16:
The `secs` in [`utime.localtime(secs)`](https://docs.micropython.org/en/latest/unix/library/utime.html?highlight=localtime#utime.localtime) is always in UTC, isn't it?

> Time Epoch: Unix port uses standard for POSIX systems epoch of 1970-01-01 00:00:00 UTC. However, embedded ports use epoch of 2000-01-01 00:00:00 UTC. 
https://docs.micropython.org/en/latest/unix/library/utime.html

pfalcon, 2018-09-16:
Given that there's no support for tz in the utime module, it's up to a user to adjust input values to utime.localtime() if they want to get result in other TZ but the UTC. This is like, a pretty obvious solution. (Even if it's workaround.) For the rest, there's #2130.

lumbric, 2018-09-16:
Hm yes, as mentioned feel free to close as won't fix or duplicate of #2130. I opened a seperate ticket, because I noticed that `localtime()` behaves different in the unix port and elsewhere. I thought this might be worth fixing independently of #2130.

pfalcon, 2018-09-16:
That's because unix port has means to deal with it in the OS, while baremetal ports currently don't have means to deal with it at all. And the argument in #2130 is about whether to make it fully consistent across ports (both getting and setting), or follow CPython way where it will be idiosyncratic per port.

lumbric, 2018-09-24:
I don't really understand the details, but if I understand @pfalcon correctly, this can be closed as duplicate of #2130.

closed 2018-09-24 by lumbric