micropython/micropython #4147: pyboard: baudrate of SPI interface is incorrect

peterhinch, 2018-09-16:
This script:
```python
import machine
import utime
spi = machine.SPI(1)
spi.init(baudrate=10000000)
buf = bytearray((0x0f for _ in range(100)))
while True:
    spi.write(buf)
    utime.sleep_ms(100)
```
was run, varying only the baudrate. Clock and data signals were checked on a scope. Outcome for the clock signal:

| Set value | Measured | Ratio |
|:-------------:|:--------------:|:----:|
|   2MHz |  1.3MHz | 0.65 |
|  4MHz  | 2.62MHz | 0.65 |
| 8MHz | 5.26MHz | 0.65 |
| 10MHz | 5.3MHz | 0.53 |
| 20MHz |10.48MHz | 0.52 |

pyb produces the same result (tested only at 2MHz nominal). Target:

```
MicroPython v1.9.4-479-g828f771-dirty on 2018-09-11; PYBv1.1 with STM32F405RG
```

dpgeorge, 2018-09-16:
When using `machine.SPI`, can you please print out the spi object (`print(spi)` after calling init), and also `print(pyb.freq())`?  The hardware SPI on STM32 parts is quite limited in the frequencies it can do and it may be that it's rounding them to the nearest supported baudrate (that's why 47550ef2cd4add384781c390ac8073beb71be1a1 was done, although that only helps with F7 MCUs).

peterhinch, 2018-09-16:
The SPI bus reports the actual frequencies I measured.
```python
import machine
import utime
import pyb
spi = machine.SPI(1)
spi.init(baudrate=2000000)
print(spi, pyb.freq())
buf = bytearray((0x0f for _ in range(100)))
while True:
    spi.write(buf)
    utime.sleep_ms(100)
```
| baud | Outcome |
|:-----:|:-----:|
| 2MHz | SPI(1, baudrate=1312500, polarity=0, phase=0, bits=8) (168000000, 168000000, 42000000, 84000000) |
|  4MHz | SPI(1, baudrate=2625000, polarity=0, phase=0, bits=8) (168000000, 168000000, 42000000, 84000000) |
| 8MHz | SPI(1, baudrate=5250000, polarity=0, phase=0, bits=8) (168000000, 168000000, 42000000, 84000000) |
| 10Mhz | SPI(1, baudrate=5250000, polarity=0, phase=0, bits=8) (168000000, 168000000, 42000000, 84000000) |
| 20MHz | SPI(1, baudrate=10500000, polarity=0, phase=0, bits=8) (168000000, 168000000, 42000000, 84000000) |

So it seems to be a hardware limit. I was trying to run at 20MHz and it looks like the nearest I can get (without exceeding 20MHz) is 13.125. But to get that I have to specify over 20MHz. This is counterintuitive: why does calling up 20MHz give 10.48MHz when the hardware can achieve 13.125MHz?

I could amend the docs to indicate that available rates are limited.

closed 2018-09-20 by peterhinch

dpgeorge, 2018-09-20:
> So it seems to be a hardware limit.

Yes it is a hardware limit.

> I could amend the docs to indicate that available rates are limited.

That would be good, but finding the right place to put that info is the tricky bit...