micropython/micropython #4155: stm32: usb vcp hangs on large transfers due to WFI

andrewleech, 2018-09-17:
We're transferring large amounts of data over the virtual serial port(s) such as images ~2MB and it regularly hangs part way through. 
This is incredibly common for us with large transfer buffers, sent with `vcp.send(buf, timeout=5000)` (unbuffered).
Splitting the buffer into small chunks (250 bytes) and looping doesn't hit the same issue, and the buffered `.write()` function doesn't seem to be really affected either.

I've traced the issue to the `__WFI();` calls in https://github.com/micropython/micropython/blob/master/ports/stm32/usbd_cdc_interface.c

If I comment them out the problem is completely fixed.

I suspect it's a race condition between the while loop conditionals checking if there's data ready to be handled, when a couple of operations before WFI, if the interrupt and incoming data occurs after the while check but before the wfi the data wont get handled meaning there wont be any follow up interrupt to trigger the exit from wfi.

If I send some data to the board (eg type something into serial terminal) it becomes unstuck, and the transfer continues.  This also supports my theory that WFI just gets stuck without any kind of timeout.

While I'm a big fan of conserving power and resources whenever you can, the WFI's in the VCP functions are only hit when it's in the middle of an active USB transfer, are they really valuable enough to keep in? 

Assuming I'm correct in my race condition theory, I'm not too sure how to resolve it, for now I'm running in a branch with all the WFI's disabled and it appears to be very stable.

andrewleech, 2018-09-17:
Looking around the code more, I don't know why the systick interrupt doesn't unstick this automatically, it should force an exit from wfi regularly even if the usb doesn't.

I actually first found this issue about a month ago and have tried re-enabling WFI a few times to check if the problem still exists and hit the symptoms immediately, so it really does seem to be repeatable for us.

dpgeorge, 2018-09-17:
Are you able to provide code that exhibits the behaviour on an officially supported board (one in the stm32/boards/ directory)?  Running a simple test on a PYBv1.0 sending 2MiB at a time (at a rate of 341kbytes/s) doesn't show any issues; test code is:
```python
import time, uctypes, pyb 
usb = pyb.USB_VCP()
buf = uctypes.bytearray_at(0x90000000, 2 * 1024 * 1024)
while 1:
    t0 = time.ticks_ms()
    usb.send(buf, timeout=5000)
    t1 = time.ticks_ms()
    dt = time.ticks_diff(t1, t0) 
    print(len(buf) / dt, 'kb/s')
    pyb.LED(1).toggle()
```

andrewleech, 2018-09-17:
Yep, sorry, I really should have thought of making a simple repro case first myself.

I haven't been able to reproduce it on my 429 Disco board yet, I'll put more work into narrowing down something that's easier to set up than a full imaging platform to demonstrate the issue.

andrewleech, 2018-09-17:
I actually can't repro it at all now after rebasing all my development branch to head, previously failing things seem to be working fine now. 
Sorry about the noise, I suspect something else that's changed recently has fixed the issue without me noticing. I'll not bother pushing a PR for this unless it comes back up for me later!

closed 2018-09-17 by andrewleech

Lak4CYUT, 2021-05-29:
Actually I encounter a similar issue when I develop OpenMV's driver. 
My entire system seems hang on __WFI(), it is really weird.

It is happen when I try to printf some log messages in DCMI callback interrupt. 
Do you guys have any idea about that?