micropython/micropython #4166: Non deterministic write times as SD card fills up

rhubarbdog, 2018-09-19:
I have a pyboard 1.1 with an 8gb SD card. There is a SKIPSD file on the flash. I've filled the SD card with mixture of 1 block files and bigger files (.WAV) and linux iso images. (It takes too long to write multiple 1 block files to fill the disk.)  
From a full SD card i delete 30 random 1 block files.
I then run this program which writes 256 bytes every 0.1 seconds. most writes take 0.1 seconds every so often the program pauses for approximately 4 seconds.
[bad_sdcard.py.txt](https://github.com/micropython/micropython/files/2398913/bad_sdcard.py.txt)
[results.txt](https://github.com/micropython/micropython/files/2398915/results.txt)

adritium, 2018-09-20:
Simplify the code: Remove all lines that have nothing to do with writing 256 bytes.

rhubarbdog, 2018-09-20:
@adritium sorry for leaving all that chaff in my listing. I initially thought the problem was in the creation of the `boot-file.txt` all excess code removed here's a better listing
[bad_sdcard.py.txt](https://github.com/micropython/micropython/files/2399415/bad_sdcard.py.txt)
@chrismas9 I've tried this using my laptop with a python3 version on `bad_sdcard.py` using `time.perf_counter()` rather than micropython `time.ticks_us()`. All writes take about 0.1000 seconds some (5?) take 1 or 2ms longer and another one or two takes 20ms longer.

adritium, 2018-09-20:
There’s still way too much going on to figure out the problem. 

(1) get the time
(2) write something to SD card
(3) get time again and print diff to REPL. 

This way you’re only timing the actual write and you’re only writing what you time.

rhubarbdog, 2018-09-20:
@adritium your mis-reading the code
on the first loop the time is only the write time for 4 lines of 64 bytes approx 5-10ms
all subsequent loops the time is the write time and the `pyb.delay(100)` approx 105ms
there are 1800 lines in `data_file.txt` of which only 8 exhibit the behaviour of taking approximately 4 seconds to complete the loop

adritium, 2018-09-20:
That wasn't the point of my comment.

The point was

> This way you’re only timing the actual write and you’re only writing what you time.

adritium, 2018-09-20:
> If all the delays are in the card they should be similar on a PC.

Could the PC have cache enabled on the PC?

peterhinch, 2018-09-20:
> If all the delays are in the card they should be similar on a PC.

@chrismas9 I'd question that. On a PC there is an OS between Python and the card. Writes are buffered (which is why you have to "safely eject" the card). I'd expect times to be fast regardless of underlying delays in the card interface.

chrismas9, 2018-09-20:
@peterhinch. I had forgotten about caching because write caching for removable drives is disabled for Windows. If Linux has write caching enabled by default it would need to be disabled.

rhubarbdog, 2018-09-20:
@chrismas9 as per @peterhinch 's comments 
I believe to have tried it using python3 with write cache disabled.  
I tried `sudo hdparm -i /dev/sdb` (my laptops sdcard reader) the output was inconclusive (due to a bug? in hdparm with sdcards).
trying `sudo hdparam -i /dev/sda` (my laptops hard drive), yields write cache enabled.
I edited `/etc/hdparm.conf` to disable write cache and rebooted
`sudo hdparm -i /dev/sda` yields write cache disabled.
I free some random blocks and run the python3 version of `bad_sdcard.py` there are no times in excess of 0.12 seconds

adritium, 2018-09-20:
You’re changing too many variables: OS, Python implementation, hardware. 

You have to change ONE at a time.

peterhinch, 2018-09-21:
I'm no expert but I avoid letting any filesystem get more than 90% full. With a FAT variant I'd be even more cautious. Is this really worth pursuing? My approach would be to put a note in the docs, with a code sample showing how an application might check free space.

rhubarbdog, 2018-09-24:
I agree with @peterhinch about leaving 10-20% free.  It was a naive program which led me to find there were issues as the disk filled. Because leaving 1GB on an 8GB disk seems a horrible waste, I thought I'd find the cut off for long non deterministic write times. In one attempt I think one of my files was truncated, but can't proove it.  I have evidence of damaged file names in my directory. All names should be of the form "block%02d.txt" but there is a wide variation of damage.
```
os.listdir()
['block01.txt', 'block02.txt', 'block03.txt', 'block04.txt', 'block05.txt', 'block06.txt', 'block07.txt', 'block08.txt', 'block09.txt', 'block10.txt', 'block12.txt', 'block13.txt', 'block14.txt', 'block16.txt', 'block17.txt', 'block18.txt', 'block19.txt', 'block20.txt', 'block21.txt', 'block22.txt', 'block23.txt', 'block24.txt', 'block25.txt', 'block26.txt', 'block27.txt', 'block28.txt', 'block29.txt', 'block30.txt', 'blkck30.txu', 'block33.txt', 'block35.txt', 'blgck37.txt', 'block38.tyt', 'b\x0cock3;.txt', 'block40.txt', 'block41.txt', 'block42.txt', 'elock43.txt', 'block44.pxt', 'block45.txt', 'block46', 'block48.txt', 'block4y.txt', 'BLOCk50.txt', 'block5.txt', 'block52.txt', 'block53$.txt', 'block54.txt', 'block56.txt', 'clock57$.tht', 'block58.txt', 'clock59.tzt', 'block60.tx\\', 'block6\x11.txt', 'bloqk>2.txt', 'block63.txt']
```

causing the line 
`os.remove(random.choice(os.listdir()))` to raise an OSError exception ENOENT

adritium, 2018-09-24:
Go in stackoverflow and ask how much you should leave free based on your file system. 

Should get an answer very fast.

rhubarbdog, 2018-09-25:
I'd thought i'd have a quick look at write times on a disk that was empty and half empty.
There are no super long writes, but the times are a bit erratic. 90% of the time a block of data is written in 35ms, but the other 10% of the time the time it takes 100 -> 250ms.
As for full disks i was achieving a mean time to write a block of 53ms. An 8GB disk has nearly 2 million 4KB blocks which implies it takes one and a half days to write 8GB.  Will any real world applications get anywhere near 80% of an SD card?

jonnor, 2024-07-27:
Proposing to close this. This is most likely due to the SD card drivers, which MicroPython cannot do much about.

closed 2024-07-28 by mattytrentini