micropython/micropython #4170: ESP32 support for I2S

nickzoic, 2018-09-20:
This previously existed as https://github.com/micropython/micropython-esp32/issues/185 but I'm replicating it here.  The ESP32 has I2S hardware so it'd be nice to support it!

References:
* [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/i2s.html)
* [MicroPython I2S API](https://github.com/micropython/micropython/wiki/Hardware-API#the-i2s-class)

nickzoic, 2018-09-20:
(I've got some hardware in the junkbox but have had no time to make any progress on this.  If you've got the time and inclination, go for it!)

AidanTek, 2018-09-20:
I am very keen to see this happen and will help in any way I can - implementation is above my current skill level but I would very much like to collaborate and learn.

I have a couple of I2S audio codecs - PCM1808 for stereo in, PCM5102 for stereo out and also have some Teensy Audio Adaptors with the SGTL5000 (Line In/Out and headphones out). The PCM1808 and PCM5102 are cheap (available on Mouser) and can be hand soldered if you are comfortable with easier surface mount soldering, modules are also available on eBay and other places. The audio quality is very good!

miketeachman, 2019-02-10:
As part of an air and noise pollution sensor project (https://hackaday.io/project/162059-street-sense) I needed to support an I2S microphone using the ESP32 port. Using the ESP-IDF I implemented support for both I2S Master Rx and Tx . I spent a good deal of effort to bring the C code from a "just works hobbyist" quality to "industrialized" quality.

I put in a PR to mainline uPy:
[micropython/micropython#4471](https://github.com/micropython/micropython/pull/4471)

Detailed class usage documentation and working uPy examples for popular microphone and audio DAC breakout boards (including the PCM5102) is here:
https://github.com/miketeachman/micropython-esp32-i2s-examples

Feedback and comments welcome at the PR or at the documentation repo

Mike

dpgeorge, 2022-01-23:
I2S is supported since 8a5bfe44a5b686b479ef0638e25fe968099f2a28

closed 2022-01-23 by dpgeorge