micropython/micropython #4172: Create builtin to clear module from RAM

adritium, 2018-09-20:
[To remove a module from RAM](https://forum.micropython.org/viewtopic.php?f=2&t=2639), you have to:

```Py
del themod
if themod.__name__ in sys.modules:
    del sys.modules[themod.__name__]
gc.collect()
```

Maybe this code should be a builtin function since it saving RAM is a common operation.

Of course this could technically fail if you have another reference to the module . . .

jpochyla, 2018-09-25:
This is not really enough, you also need to remove the reference from the parent package.

adritium, 2018-09-25:
Ok whatever. 

It’d be useful to have this ability.

pmp-p, 2018-09-30:
i personnally do "with" blocks and an import function named uses ( i love pascal too ;) )
with uses('themod'):
    themod.do_something()
where __exit__ will clear the mod if it was not imported before __enter__, and refcounting when nesting.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge